RROrg/rr #4434: Updating to 7.2.2 and getting "Failed to install file. The file is probably corrupted"

cri5py, 2024-11-20:
请填写以下信息.  
Please fill in the following information.  (FULL LOG ATTACHED IN TXT FILE)

Install ENV: (You can find it in the boot interface.)  
 * DMI:  Gigabyte Technology Co., Ltd. Z68M-D2H/Z68M-D2H, BIOS F6 02/21/2012
 * CPU:  CPU0: Intel(R) Core(TM) i7-3770 CPU @ 3.40GHz
 * NIC:  (pid & vid) id=0x5575

RR version: (You can find it in the update menu.)  
 * RR:  rr-24.11.11.img
 * addons:  
 * modules:  Modules linked in: r8168(OE+) pgdrv(O) r8169_lk(OE) mii(OE) evdev(OE) r8168_tx(OE) e1000e(OE+) button(OE) processor(+) xhci_pci xhci_hcd uhci_hcd ehci_pci ehci_hcd usbcore usb_common mv14xx(O)

 * lkms:  

DSM: 
 * model:  DS3622xs+
 * version:  (Current) DSM 7.2-64570 Update 3, Failing when trying to upgrade to 7.2.2

Issue:  

When trying to upload the PAT file, it fails at 58% and gives the error: "Failed to install file. The file is probably corrupted". I tried several times but it always fails. 

logs:  

`Nov 20 16:24:11  synoupgrade: ninstaller.c:140 Mount partion /dev/md0 /tmpRoot
Nov 20 16:24:11  kernel: [   32.246716] EXT4-fs (md0): couldn't mount as ext3 due to feature incompatibilities
Nov 20 16:24:11  kernel: [   32.258009] EXT4-fs (md0): mounted filesystem with ordered data mode. Opts: (null)
Nov 20 16:24:11  synoupgrade: ninstaller.c:169 umount partition /tmpRoot
Nov 20 16:24:11  synoupgrade: ninstaller.c:140 Mount partion /dev/md0 /tmpRoot
Nov 20 16:24:11  kernel: [   32.285464] EXT4-fs (md0): couldn't mount as ext3 due to feature incompatibilities
Nov 20 16:24:11  kernel: [   32.298942] EXT4-fs (md0): mounted filesystem with ordered data mode. Opts: (null)
Nov 20 16:24:11  synoupgrade: ninstaller.c:511 fail to get admin port from /tmpRoot/etc/synoinfo.conf... try another way
Nov 20 16:24:11  synoupgrade: ninstaller.c:892 flash version is newer than disk version
Nov 20 16:24:11  synoupgrade: ninstaller.c:1307 SYSTEM_NORMAL, [Recoverable=0]
Nov 20 16:24:11  synoupgrade: ninstaller.c:169 umount partition /tmpRoot
Nov 20 16:24:11  synoupgrade: ninstaller.c:1222(FillUpgradeVolumeInfo): gszUpgradeVolDev = /dev/md0 
Nov 20 16:24:11  synoupgrade: ninstaller.c:1223(FillUpgradeVolumeInfo): gszUpgradeVolMnt = /tmpData 
Nov 20 16:24:11  synoupgrade: ninstaller.c:1331 gblSupportRaid: 1, gSysStatus: 1, gblCreateDataVol: 0, gblSystemRecoverable: 0
Nov 20 16:24:13  synodiskupdatehclport: misc/misc_ds_info_get.c:126 Failed to get DS SN
Nov 20 16:24:13  synodiskupdatehclport: synodiskupdatehclport.c:63 Failed to get ds sn
Nov 20 16:24:13  synodiskupdatehclport: synodiskupdatehclport.c:104 Fail to get online server info
Nov 20 16:24:13  synodiskupdatehclport: synodiskupdatehclport.c:318 Fail to get hcl info from online server
Nov 20 16:24:15  synoupgrade: autocleanup/mount.c:19 run '/usr/bin/mount /dev/md0 /_disk'
Nov 20 16:24:15  kernel: [   35.564945] EXT4-fs (md0): couldn't mount as ext3 due to feature incompatibilities
Nov 20 16:24:15  synoupgrade: autocleanup/mount.c:27 mount's exit status is zero
Nov 20 16:24:15  synoupgrade: get_disk_dsm_version.c:59 try reading disk dsm's version at file '/_disk//etc.defaults/VERSION' of key 'buildnumber'
Nov 20 16:24:15  kernel: [   35.598616] EXT4-fs (md0): mounted filesystem with ordered data mode. Opts: (null)
Nov 20 16:24:15  synoupgrade: autocleanup/mount.c:44 run '/usr/bin/umount -l /_disk'
Nov 20 16:24:15  synoupgrade: autocleanup/mount.c:54 umount's exit status is zero
Nov 20 16:24:15  synoupgrade: autocleanup/directory.c:38 remove directory /_disk, result is 0
Nov 20 16:24:15  synodiskport: disk/disk_rule_junior.c:245 Failed to get target on WD80EDAZ-11TA3A0 81.00A81 72806.0
Nov 20 16:24:15  synodiskport: disk/disk_rule_junior.c:245 Failed to get target on HUS728T8TALE6L4 V8GNW410 72806.0
Nov 20 16:24:15  synodiskport: disk/disk_rule_junior.c:245 Failed to get target on DT01ACA200 MX4OABS0 72806.0
Nov 20 16:24:15  synodiskport: disk/disk_rule_junior.c:245 Failed to get target on HUS728T8TALE6L4 V8GNW410 72806.0
Nov 20 16:24:15  synodiskport: disk/disk_rule_junior.c:245 Failed to get target on HUS728T8TALE6L4 V8GNW410 72806.0
Nov 20 16:24:17  synoagentregisterd: synoagentregisterd.c:149 Generating register key...
Nov 20 16:24:17  synoagentregisterd: synoagentregisterd.c:163 Generated register key [D08B062C.9C0F.464E.8B4B.7E43D030219C].
Nov 20 16:24:23  synodiskport: disk/disk_rule_junior.c:245 Failed to get target on WD80EDAZ-11TA3A0 81.00A81 72806.0
Nov 20 16:24:23  synodiskport: disk/disk_rule_junior.c:245 Failed to get target on HUS728T8TALE6L4 V8GNW410 72806.0
Nov 20 16:24:23  synodiskport: disk/disk_rule_junior.c:245 Failed to get target on DT01ACA200 MX4OABS0 72806.0
Nov 20 16:24:23  synodiskport: disk/disk_rule_junior.c:245 Failed to get target on HUS728T8TALE6L4 V8GNW410 72806.0
Nov 20 16:24:23  synodiskport: disk/disk_rule_junior.c:245 Failed to get target on HUS728T8TALE6L4 V8GNW410 72806.0
Nov 20 16:24:25  synodiskport: disk/disk_rule_junior.c:245 Failed to get target on WD80EDAZ-11TA3A0 81.00A81 72806.0
Nov 20 16:24:25  synodiskport: disk/disk_rule_junior.c:245 Failed to get target on HUS728T8TALE6L4 V8GNW410 72806.0
Nov 20 16:24:25  synodiskport: disk/disk_rule_junior.c:245 Failed to get target on DT01ACA200 MX4OABS0 72806.0
Nov 20 16:24:25  synodiskport: disk/disk_rule_junior.c:245 Failed to get target on HUS728T8TALE6L4 V8GNW410 72806.0
Nov 20 16:24:25  synodiskport: disk/disk_rule_junior.c:245 Failed to get target on HUS728T8TALE6L4 V8GNW410 72806.0
Nov 20 16:24:25  install.cgi: install.c:571 Unknown param localinstallreq=false
Nov 20 16:24:25  install.cgi: install.c:571 Unknown param _dc=1732119865916
[... 47 lines cut ...]
Jan  1 00:00:30  synodiskport: disk/disk_rule_junior.c:245 Failed to get target on WD80EDAZ-11TA3A0 81.00A81 72806.0
Jan  1 00:00:30  synodiskport: disk/disk_rule_junior.c:245 Failed to get target on HUS728T8TALE6L4 V8GNW410 72806.0
Jan  1 00:00:30  synodiskport: disk/disk_rule_junior.c:245 Failed to get target on DT01ACA200 MX4OABS0 72806.0
Jan  1 00:00:30  synodiskport: disk/disk_rule_junior.c:245 Failed to get target on HUS728T8TALE6L4 V8GNW410 72806.0
Jan  1 00:00:30  synodiskport: disk/disk_rule_junior.c:245 Failed to get target on HUS728T8TALE6L4 V8GNW410 72806.0
Jan  1 00:00:38  kernel: [  119.743804] EXT4-fs (md0): couldn't mount as ext3 due to feature incompatibilities
Jan  1 00:00:38  kernel: [  119.785905] EXT4-fs (md0): mounted filesystem with ordered data mode. Opts: (null)
Jan  1 00:00:38  synodiskport: disk/disk_rule_junior.c:245 Failed to get target on WD80EDAZ-11TA3A0 81.00A81 72806.0
Jan  1 00:00:38  synodiskport: disk/disk_rule_junior.c:245 Failed to get target on HUS728T8TALE6L4 V8GNW410 72806.0
Jan  1 00:00:38  synodiskport: disk/disk_rule_junior.c:245 Failed to get target on DT01ACA200 MX4OABS0 72806.0
Jan  1 00:00:38  synodiskport: disk/disk_rule_junior.c:245 Failed to get target on HUS728T8TALE6L4 V8GNW410 72806.0
Jan  1 00:00:38  synodiskport: disk/disk_rule_junior.c:245 Failed to get target on HUS728T8TALE6L4 V8GNW410 72806.0
Jan  1 00:00:46  synodiskport: disk/disk_rule_junior.c:245 Failed to get target on WD80EDAZ-11TA3A0 81.00A81 72806.0
Jan  1 00:00:46  synodiskport: disk/disk_rule_junior.c:245 Failed to get target on HUS728T8TALE6L4 V8GNW410 72806.0
...

(请先看一下#173、#175、#226 的内容)  
[log.txt](https://github.com/user-attachments/files/17836391/log.txt)

(Plz review the content of #173, #175, #226 first)  
...

(如果你只是说 XXX 不能用, 什么详细信息也不提供, 我也只能说感谢你的反馈.)  
(If you just say XXX doesn't work without providing any details, I can only say thank you for your feedback.)  
...

i0xi-1024, 2024-11-21:
@cri5py  Sounds like you are using the wrong upgrade file. You need to upgrade to 7.2.2-72806 first, then upgrade to 7.2.2-72806 Update 1, and check if your pat file is 300M+?

cri5py, 2024-11-21:
Hey Thanks for looking into it. But I double-checked and redownloaded the file directly from Syno. It fails as before. Looking over the logs I wonder if it has something to do with the serial number. 

Nov 21 21:18:10  kernel: [    6.544783] Got empty serial number. Generate serial number from product.

[log2.txt](https://github.com/user-attachments/files/17852178/log2.txt)

cri5py, 2024-11-21:
Seems that I had to leave it at 7.1. Then I ran made space for the update. Uploaded the new DSM update via the DSM web interface and it worked. I used to try to do it via the re-install which made it fail.  Now it works, although my networking scripts and firewall rules are messed up, I was able to detele them and recover the defaults. Thanks for all the help.

closed 2024-11-21 by cri5py