RROrg/rr #4473: 【HELP】安装一直卡在 get pat data这一步，如何解决？ 😄

myworksfun, 2024-11-25:
rr版本： 24.11.12
cpu：N5150
ENV: VM

[image]

wjz304, 2024-11-25:
[image]

closed 2024-11-25 by wjz304

myworksfun, 2024-11-25:
> ![image](https://private-user-images.githubusercontent.com/5615843/389433377-f51c3ddf-d445-4dbb-9df4-21046789e016.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzI1MjIzNzAsIm5iZiI6MTczMjUyMjA3MCwicGF0aCI6Ii81NjE1ODQzLzM4OTQzMzM3Ny1mNTFjM2RkZi1kNDQ1LTRkYmItOWRmNC0yMTA0Njc4OWUwMTYucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MTEyNSUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDExMjVUMDgwNzUwWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9MDNjN2M0Y2ZmMzBkMjAxOWQ5OWNkMDk1MjRkZGQwZTJiNWVhMDJiNmQ3YjQ3YTA5OTdjYmYzOTE5ODVmZTE3YiZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.-9ZgP69Jf72Aw7_hVyxqAsAg5TF5o6mB0vnL24KK6xI)

### OK,GOT IT , THANK YOU.