RROrg/rr #4509: 最新版rr无法创建必需的任务。

saineiru, 2024-11-27:
请填写以下信息.  
Please fill in the following information.  

Install ENV: (You can find it in the boot interface.)  
 * DMI:  ASUS B360M PLUS
 * CPU:  i5 9600
 * NIC:  (pid & vid)

RR version: (You can find it in the update menu.)  
 * RR:  2.0-79
 * addons:  
 * modules:  
 * lkms:  

DSM: 
 * model:  SA6400
 * version:  DSM 7.2.1-69057 Update 5

Issue:  每次开打应用都提示需要创建两个任务“RunRrUpdate”和“ApplyRRConfig”，每次提示创建成功后需要重启应用，但是任务实际上并没有被创建，打开应用后依旧提示需要创建必需任务....
[image]

logs:  

(## 因为 log中存在 SN/MAC 等一些敏感信息, 当提供完整文件时请自行抹除他们, 当然你也可以发送到我的邮箱. ##)  
(## Because the log contains some sensitive information such as SN/MAC, please delete them when providing the complete file. Of course, you can also send it to my email. ##)    
...

(请先看一下#173、#175、#226 的内容)  
(Plz review the content of #173, #175, #226 first)  
...

(如果你只是说 XXX 不能用, 什么详细信息也不提供, 我也只能说感谢你的反馈.)  
(If you just say XXX doesn't work without providing any details, I can only say thank you for your feedback.)  
...

CaesarLsky, 2025-02-15:
一样的问题，持续很久都没有解决😂

wjz304, 2025-03-11:
https://github.com/RROrg/rr/issues/7108

closed 2025-03-11 by wjz304