RROrg/rr #4510: No booting Loader

zlodeychik, 2024-11-28:
Good afternoon!
I can't load into Loader.
After booting from USB, the main Loader window appears.
[image]
Select Configure loader.
Nothing happens next. The cursor does not blink after "Booting...". The USB flash drive is also not flashing.
[image]
I tried the old Loader arpl-1.1-beta 2 it runs. I tried to change the USB flash drive and different versions of RR 24.11.13, 24.11.12, 24.11.11do not work. Memtest86+ doesn't start either, stops on Booting.

Please fill in the following information.  

 * DMI: Fujitsu Esprimo c720 MB:  FUJITSU D3224-A1, Intel Q85 chipset
 * CPU: QuadCore Intel Xeon E3-1265L v3, 2500 MHz, 16GB RAM
 * NIC:  Intel(R) Ethernet Connection I217-V PCI\VEN_8086&DEV_153B&SUBSYS_11ED1734&REV_04

RR version: 24.11.13, 24.11.12, 24.11.11

wjz304, 2024-11-28:
How long did you wait after it displayed "Booting..."?

zlodeychik, 2024-11-28:
About 30 minutes each time. arpl-1.1-beta the line "Booting device USB" appears after 5 seconds

zlodeychik, 2024-11-28:
I tried all the USB ports in the PC. 
The keyboard stops working after "Booting..."

wjz304, 2024-11-28:
[image]
Select "configure loader" and click "e"

[image]
Move the cursor here and enter "console=tty2",

[image]
click "F10"

Observe the screen output

zlodeychik, 2024-11-28:
Thank you. I'll try.

zlodeychik, 2024-11-28:
Nothing has changed.
[image]

wjz304, 2024-11-28:
Then there is no way to investigate, let's use arpl to boot first,

zlodeychik, 2024-11-28:
arpl is working.
[image]
[image]
[image]

wjz304, 2025-03-11:
无法调查原因

closed 2025-03-11 by wjz304