RROrg/rr #4526: 关于RR引导下端口能否建立LACP -I am writing to inquire about the possibility of establishing LACP on the network port under RR boot.

zhangpinzheng, 2024-12-02:
I have a query regarding the use of Link Aggregation Control Protocol (LACP) on Synology devices. Could you kindly confirm whether RR Boot can utilise LACP? I have encountered an issue where RR Boot is unable to obtain an IP address correctly in switch LACP mode during the boot process.
我有一个问题我的Synology使用了两个lacp保证数据传输稳定，rr引导可以使用lacp吗？因为我发现在启动过程中它在交换机lacp 模式下不能正常获取IP

wjz304, 2024-12-02:
RR 是 RR，DSM 是DSM，RR 支不支持都不影响 DSM

wjz304, 2024-12-02:
[image]

zhangpinzheng, 2024-12-02:
> RR 是 RR，DSM 是DSM，RR 支不支持都不影响 DSM

我明白，我的意思是我交换机建立LACP后RR引导无法获取DHCP,我有点不知道怎么在RR引导里面建立bond组，让RR可以在使用端口聚合的情况下获取IP，显示dsm是正常的，我是想让rr也可以获取IP

zhangpinzheng, 2024-12-02:
> RR 是 RR，DSM 是DSM，RR 支不支持都不影响 DSM

我在研究研究，也打扰您了

wjz304, 2024-12-02:
意义不大，RR 只在启动的几秒进入获取环境信息而已

wjz304, 2024-12-03:
24.12.1  添加了 bonding

/etc/network/interfaces
```
auto bond0
iface bond0 inet static
    address 192.168.1.100
    netmask 255.255.255.0
    gateway 192.168.1.1
    bond-slaves eth0 eth1
    bond-mode 4
    bond-miimon 100
    bond-lacp-rate 1
    bond-xmit-hash-policy layer2+3

auto eth0
iface eth0 inet manual
    bond-master bond0

auto eth1
iface eth1 inet manual
    bond-master bond0
```
自己研究吧

zhangpinzheng, 2024-12-03:
> 24.12.1 添加了 bonding
>
> /etc/network/interfaces
>
> ```
> auto bond0
> iface bond0 inet static
>     address 192.168.1.100
>     netmask 255.255.255.0
>     gateway 192.168.1.1
>     bond-slaves eth0 eth1 [...]

谢谢您了

closed 2025-02-17 by wjz304

zhangpinzheng, 2025-08-03:
> 24.12.1 添加了 bonding
>
> /etc/network/interfaces
>
> ```
> auto bond0
> iface bond0 inet static
>     address 192.168.1.100
>     netmask 255.255.255.0
>     gateway 192.168.1.1
>     bond-slaves eth0 eth1 [...]
大佬怎么保存下次启动还存在这个参数