RROrg/rr #4527: 天钡wtr pro物理机安装SA6400，虚拟机安装Debian 12无法启动

smomop, 2024-12-02:
请填写以下信息.  
Please fill in the following information.  

Install ENV: (You can find it in the boot interface.)  
 * DMI:  天钡wtr pro
 * CPU:  N100
 * NIC:  (pid & vid)

RR version: (You can find it in the update menu.)  
 * RR:  24.11.13
 * addons:  nvmesystem,cpuinfo
 * modules:  
 * lkms:  

DSM: 
 * model:  
 * version:  

Issue:  
安装debian12，使用q35或者pc机型安装，磁盘3种模式也都选择了一遍，uefi启动和legacy bios也试过了，一直卡如下图，可以正常进入安装，安装后就一直卡引导
logs:  
[image]

wjz304, 2024-12-03:
N100 安装 debian-12.8.0-amd64-DVD-1.iso 不复现，

smomop, 2024-12-03:
nvmevloume这个插件会影响吗？实在找不到原因了，修改dts也会吗？

RickGanKL, 2024-12-04:
这个问题并不RR引起的,我可以确认。
这个是群晖的VMM问题。

closed 2024-12-04 by smomop

wjz304, 2024-12-05:
> wtr pro似乎是5825u，AMD虚化问题吧？

[image]
[image]