RROrg/rr #4541: BUG 24.12.1 24.11.13 nuc9引导会报out of memory，在24.11.1版本可以正常引导

meishild, 2024-12-03:
请填写以下信息.  
Please fill in the following information.  

Install ENV: (You can find it in the boot interface.)  
 * DMI:  
 * CPU:  
 * NIC:  (pid & vid)

RR version: (You can find it in the update menu.)  
 * RR:  
 * addons:  
 * modules:  
 * lkms:  

DSM: 
 * model:  
 * version:  

Issue:  

logs:  

(## 因为 log中存在 SN/MAC 等一些敏感信息, 当提供完整文件时请自行抹除他们, 当然你也可以发送到我的邮箱. ##)  
(## Because the log contains some sensitive information such as SN/MAC, please delete them when providing the complete file. Of course, you can also send it to my email. ##)    
...

(请先看一下#173、#175、#226 的内容)  
(Plz review the content of #173, #175, #226 first)  
...

(如果你只是说 XXX 不能用, 什么详细信息也不提供, 我也只能说感谢你的反馈.)  
(If you just say XXX doesn't work without providing any details, I can only say thank you for your feedback.)  
...

meishild, 2024-12-03:
没有进入系统安装就已经出错了，出现选择菜单后点确认就会出错。

wjz304, 2024-12-04:
手动改下 引导 RR1 分区 boot/grub/grub.cfg 中 搜索 nointremap  并去掉
测试并反馈

wjz304, 2024-12-04:
内存多大

meishild, 2024-12-04:
> 内存多大

32G 内存足够，rr-24.11.13 我单独下了这个也不行

wjz304, 2024-12-04:
在 nointremap 后面添加 "mem=4G "

wjz304, 2024-12-04:
或者看看bios 吧，没见过限制 initrd 大小的bios

wjz304, 2024-12-04:
不知大有什么选项

wjz304, 2024-12-04:
uefi?

wjz304, 2024-12-04:
24.11.11 报错吗

meishild, 2024-12-05:
> 24.11.11 报错吗

刚刚测试了以下可以的

wjz304, 2024-12-05:
目测你机器 对 initrd 的大小上限 256M，只能删东西了，
不过如果后期新加功能 就没办法了

wjz304, 2024-12-05:
等下个版本试下吧，不过你奶奶您最好还是确认bios是否可以解除该限制

meishild, 2024-12-05:
> 等下个版本试下吧，不过你奶奶您最好还是确认bios是否可以解除该限制

那我刷以下bios把，据说nuc9的bios坑特别多我找个最新的试试

closed 2024-12-05 by wjz304