RROrg/rr #4551: BUG 如何有效锁定DT文件的pcie地址内容

naoki66, 2024-12-05:
请填写以下信息.  
Please fill in the following information.  

Install ENV: (You can find it in the boot interface.)  
 * DMI:  海康R1
 * CPU:  n100
 * NIC:  (pid & vid) i226

RR version: (You can find it in the update menu.)  
 * RR:  v24.12.1
 * addons:  
 * modules:  
 * lkms:  

DSM: 
 * model:  SA6400
 * version:  7.2(72806)

Issue:  

由于USB失败不太准确，使用我一般采用自定义DTS文件，但是发现

当插入nvme和拔除nvme的时候，lspic获取到的sata控制的地址不一致

插入的时候   cat /sys/block/sata*/device/syno_block_info
pciepath=0000:00:1d.2,00.0
拔出变成了
pciepath=0000:00:1d.0,00.0

这就导致如果我拔除了nvme，就无法识别到sata硬盘。因为sata控制器的地址发生了变化。
同理qr下也model会导致无法固定。

特来求教如何有效锁定地址，而不受到nvme硬盘的影响。

logs:  

(## 因为 log中存在 SN/MAC 等一些敏感信息, 当提供完整文件时请自行抹除他们, 当然你也可以发送到我的邮箱. ##)  
(## Because the log contains some sensitive information such as SN/MAC, please delete them when providing the complete file. Of course, you can also send it to my email. ##)    
...

(请先看一下#173、#175、#226 的内容)  
(Plz review the content of #173, #175, #226 first)  
...

(如果你只是说 XXX 不能用, 什么详细信息也不提供, 我也只能说感谢你的反馈.)  
(If you just say XXX doesn't work without providing any details, I can only say thank you for your feedback.)  
...

wjz304, 2024-12-05:
你不热插拔 RR 不用管

naoki66, 2024-12-14:
但是USB自动识别的和物理机有点不一样，能否只定义部分内容。比如usb、sata、nvme三个地址只定义一部分？

wjz304, 2024-12-16:
[image]

naoki66, 2024-12-16:
> [image]
> 

已学习了dt文件的教程，已会操作dt文件的编写。
但是苦恼的是机器的nvme插槽插入或者不插入固态硬盘时候，sata控制器的pciepath会发生改变。

wjz304, 2025-02-17:
> > [image]
> 
> 已学习了dt文件的教程，已会操作dt文件的编写。 但是苦恼的是机器的nvme插槽插入或者不插入固态硬盘时候，sata控制器的pciepath会发生改变。

主板通道共用问题，