RROrg/rr #4552: BUG EMMC引导nvme固态的情况下会触发

naoki66, 2024-12-05:
请填写以下信息.  
Please fill in the following information.  

Install ENV: (You can find it in the boot interface.)  
 * DMI:  
 * CPU:  
 * NIC:  (pid & vid)

RR version: (You can find it in the update menu.)  
 * RR:    v24.12.1
 * addons:  
  acpid: ""
  trivial: ""
  mountloader: ""
  powersched: ""
  reboottoloader: ""
  beep: ""
  cpuinfo: ""
  expands: ""
  hdddb: ""
  maiyunda: ""
  nvmesystem: ""
  nvmevolume: ""
  sensors: ""
  setrootpw: ""
  updatenotify: ""
  storagepanel: TOWER_4_Bay 1X2
  synoconfbkp: 7 R1_bkp

 * modules:  
  8021q: ""
  8139cp: ""
  8139too: ""
  "8390": ""
  9p: ""
  9pnet: ""
  9pnet_virtio: ""
  9pnet_xen: ""
  aacraid: ""
  acpi_call: ""
  acpi-cpufreq: ""
  adm1021: ""
  adm1031: ""
  adm9240: ""
  adt7470: ""
  adt7475: ""
  aesni-intel: ""
  af_key: ""
  ah4: ""
  ah6: ""
  aic94xx: ""
  alcor_pci: ""
  alx: ""
  ansi_cprng: ""
  appletalk: ""
  aqc111: ""
  asix: ""
  async_memcpy: ""
  async_pq: ""
  async_raid6_recov: ""
  async_tx: ""
  async_xor: ""
  ata_generic: ""
  ata_piix: ""
  atkbd: ""
  atl1c: ""
  atl1e: ""
  atlantic: ""
  aufs: ""
  authencesn: ""
[... 550 lines cut ...]
 * version:  7.2(72806)

Issue:  

仅有nvme的时候偶发重启会报告丢失硬盘，必须手动点击在线重组

[image]

偶发出现固件版本无法识别，要求更新硬盘数据库。

[image]

logs:  

(## 因为 log中存在 SN/MAC 等一些敏感信息, 当提供完整文件时请自行抹除他们, 当然你也可以发送到我的邮箱. ##)  
(## Because the log contains some sensitive information such as SN/MAC, please delete them when providing the complete file. Of course, you can also send it to my email. ##)    
...

(请先看一下#173、#175、#226 的内容)  
(Plz review the content of #173, #175, #226 first)  
...

(如果你只是说 XXX 不能用, 什么详细信息也不提供, 我也只能说感谢你的反馈.)  
(If you just say XXX doesn't work without providing any details, I can only say thank you for your feedback.)  
...

naoki66, 2024-12-05:
/dts-v1/;
// version: 17
// last_comp_version: 16
// boot_cpuid_phys: 0x0

/ {
    compatible = "Synology";
    model = "synology_epyc7002_sa6400";
    version = <0x1>;
    power_limit = "0,0";
    internal_slot@1 {
        protocol_type = "sata";
        ahci {
            pcie_root = "0000:00:1d.2,00.0";
            ata_port = <0x3>;
        };
    };
    internal_slot@2 {
        protocol_type = "sata";
        ahci {
            pcie_root = "0000:00:1d.2,00.0";
            ata_port = <0x2>;
        };
    };
    internal_slot@3 {
        protocol_type = "sata";
        ahci {
            pcie_root = "0000:00:1d.2,00.0";
            ata_port = <0x1>;
        };
    };
    internal_slot@4 {
        protocol_type = "sata";
        ahci {
            pcie_root = "0000:00:1d.2,00.0";
            ata_port = <0x0>;
        };
    };
    internal_slot@5 {
        protocol_type = "sata";
        ahci {
            pcie_root = "0000:00:1d.2,00.0";
            ata_port = <0x4>;
        };
    };
    nvme_slot@1 {
        pcie_root = "0000:00:1c.0,00.0";
        port_type = "ssdcache";
    };
    nvme_slot@2 {
        pcie_root = "0000:00:1d.0,00.0";
        port_type = "ssdcache";
    };
    usb_slot@1 {
        usb2 {
            usb_port = "1-2.1";
        };
        usb3 {
            usb_port = "2-2.1";
        };
    };
    usb_slot@2 {
        usb2 {
            usb_port = "1-2.2";
        };
        usb3 {
            usb_port = "2-2.2";
        };
    };
    usb_slot@3 {
        usb2 {
            usb_port = "1-2.3";
        };
        usb3 {
            usb_port = "2-2.3";
        };
    };
    usb_slot@4 {
        usb2 {
            usb_port = "1-2.4";
        };
        usb3 {
            usb_port = "2-2.4";
        };
    };
};

naoki66, 2024-12-05:
取消自定义dt后正常，不知道问题在哪里。

wjz304, 2024-12-06:
https://github.com/RROrg/rr/releases/tag/24.12.2

closed 2024-12-07 by wjz304

naoki66, 2024-12-13:
再次报告 https://github.com/RROrg/rr/releases/tag/24.12.4版本
重启后不会重建了，但是会警告，且无法消除。
[image]
[image]

naoki66, 2024-12-13:
点击状况信息后，下拉硬盘1 硬盘2  又会恢复成为绿色 良好
[image]