RROrg/rr #4567: 请问支持在rr里格式化磁盘第一个的系统分区以实现重装系统吗？

BiFangKNT, 2024-12-07:
首先，很抱歉在此处提问，因为我没找到相关的文档或论坛以供解答或提问。

我想问的是能不能在rr里格式化磁盘第一个的系统分区，然后上传或下载dsm系统包，让它自动检测数据分区，以实现无损重装系统。也就是说，把rr当类似win pe用。

wjz304, 2024-12-07:
[image]

BiFangKNT, 2024-12-07:
> ![image](https://private-user-images.githubusercontent.com/5615843/393470176-89e66203-727d-49d3-908f-a45277b4c578.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzM1NDUxNDQsIm5iZiI6MTczMzU0NDg0NCwicGF0aCI6Ii81NjE1ODQzLzM5MzQ3MDE3Ni04OWU2NjIwMy03MjdkLTQ5ZDMtOTA4Zi1hNDUyNzdiNGM1NzgucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MTIwNyUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDEyMDdUMDQxNDA0WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9NWM3MjBmYmU0Mzk3Y2VlYWY1MWM2ZjA4YTEzZDRlODgxMWM1Y2ZhYjM0Zjc0YzRlN2UxOTQxMzY1NmI4OTNiOCZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.A0bgfWWlxZ02Mp5PyNEYkhqhdFNTb1ccQHyqYRima5E)

非常感谢！

BiFangKNT, 2024-12-07:
> ![image](https://private-user-images.githubusercontent.com/5615843/393470176-89e66203-727d-49d3-908f-a45277b4c578.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzM1NDU0ODUsIm5iZiI6MTczMzU0NTE4NSwicGF0aCI6Ii81NjE1ODQzLzM5MzQ3MDE3Ni04OWU2NjIwMy03MjdkLTQ5ZDMtOTA4Zi1hNDUyNzdiNGM1NzgucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MTIwNyUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDEyMDdUMDQxOTQ1WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9YjhkYzU5YWE5MmNhMzc2NzdhN2YyMjNjYTg3MWJlNDljNGVmMDQ2ZDk5NmNkMmY3YmNiZWU5NDA1MTM5ODMxOCZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ._XZs2RbWVhW_6HIaYe4hnxugrD1b6-n59HuI0NXM2_4)

诶等等，这项的意思难道不是格式化除引导盘之外的所有分区吗？会把我的数据分区（如`/volume1`）也格了吧？

wjz304, 2024-12-07:
可以格式化单独的分区，不过你如果不会操作 还是说不建议操作

BiFangKNT, 2024-12-07:
> 可以格式化单独的分区，不过你如果不会操作 还是说不建议操作

非常感谢！到时候要用到了我再看看，我现在是重启系统就可能崩了，所以到时候要是不得不重启的话，应该需要无损重装。因为之前不小心手误把系统一部分权限搞崩了，数量太大也不太可能手动改回去🤣

closed 2024-12-07 by BiFangKNT