RROrg/rr #4582: 编译完成后启动，长时间黑屏死机，网络不上线，如何抓取log

wiseman815, 2024-12-09:
最新引导默认插件已经有dbgutils插件，按指南编译完成，启动，等10分钟仍然卡死，强制关机后进入RR菜单，选择上报问题，提示无Log日志

wiseman815, 2024-12-09:
插件已经添加，没有log文件

wiseman815, 2024-12-09:
DMI:GMKtec NucBox G3/GMKtec, BIOS GMK_G3 10/21/2023
CPU:Intel(R) N100
NIC: (pid & vid)
RR version: 24.12.3

RR:
addons:HDDdb、NvmeSystem、NvmeVolume
modules:
lkms:dev、test、prod均试过
DSM:SA6400(epyc7002)
[image]

wiseman815, 2024-12-09:
查看启动盘分区，似乎没有日志的mnt/p1路径
[image]

wjz304, 2024-12-09:
[image]
[image]

closed 2024-12-09 by wjz304

wiseman815, 2024-12-09:
> ![image](https://private-user-images.githubusercontent.com/5615843/393810208-5f823aab-8218-4e0a-9685-70fccec21983.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzM3NDU4NzAsIm5iZiI6MTczMzc0NTU3MCwicGF0aCI6Ii81NjE1ODQzLzM5MzgxMDIwOC01ZjgyM2FhYi04MjE4LTRlMGEtOTY4NS03MGZjY2VjMjE5ODMucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MTIwOSUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDEyMDlUMTE1OTMwWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9YmY5MDFjZWM4ODIwYWQ4MWMyNThlMWZiNzQwNzdlZjI5OWMyMmVkYTZjNWNhMjQzYmQxZThmNWZhYWFhMTI1NCZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.tewu5ekrf_s9Na6G-YfbIh_9Rod1cEvNXGu9oPyz24o) ![image](https://private-user-images.githubusercontent.com/5615843/393810264-136cc5e9-ca30-4933-bd56-a68190a9c5d4.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzM3NDU4NzAsIm5iZiI6MTczMzc0NTU3MCwicGF0aCI6Ii81NjE1ODQzLzM5MzgxMDI2NC0xMzZjYzVlOS1jYTMwLTQ5MzMtYmQ1Ni1hNjgxOTBhOWM1ZDQucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MTIwOSUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDEyMDlUMTE1OTMwWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9NWJkZTA0MGRjYTU3YjQzMzI4OWIwYmMwMWJjZmVjODkwZjZhOTJlMTBkNTdhMjUwNDFlODkyZDUzYmFjMDkzMSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.Hm3QCWYv0iFjs3kfNIgSxbS9fGnjP-HzcE3B-6EY99c)

这就是我😁😁