RROrg/rr #4588: BUG models not showing up. Stuck on main menu

easante22, 2024-12-09:
请填写以下信息.  
Please fill in the following information.  

Install ENV: (You can find it in the boot interface.)  
 * DMI:  
 * CPU:  
 * NIC:  (pid & vid)

RR version: (You can find it in the update menu.)  
 * RR:  v24.12.3
 * addons:  
 * modules:  
 * lkms:  

DSM: 
 * model:  sa6400
 * version:  7.1/7.2

Issue:  cannot access model menu , tries to load getting models ,
 than cycles back to main menu hangs and choose a model. 
[image]
![IMG_4083](https://github.com/user-
attachments/assets/ae0dddb0-716c-45f6-912c-c5ef4a0c9ab2)
[image]

logs:  

(## 因为 log中存在 SN/MAC 等一些敏感信息, 当提供完整文件时请自行抹除他们, 当然你也可以发送到我的邮箱. ##)  
(## Because the log contains some sensitive information such as SN/MAC, please delete them when providing the complete file. Of course, you can also send it to my email. ##)    
...

(请先看一下#173、#175、#226 的内容)  
(Plz review the content of #173, #175, #226 first)  
...

(如果你只是说 XXX 不能用, 什么详细信息也不提供, 我也只能说感谢你的反馈.)  
(If you just say XXX doesn't work without providing any details, I can only say thank you for your feedback.)  
...

wjz304, 2024-12-10:
If you cannot connect to the Internet, please build a pre-compiled bootloader through [RR-CUSTOM](https://rrorg.github.io/rr/).

easante22, 2024-12-10:
Thanks, all good, I worked around it. I was  using it to build a vm in unraid. Had  create the bootloader on a barematal machine. Than create an iso of the usb bootloader. Than run the new iso in unraid. Thanks again. Appreciate the work you do. 👍

closed 2024-12-12 by wjz304