RROrg/rr #4600: RAID on Adaptec isn't work

eridanletalis, 2024-12-11:
Install ENV: (You can find it in the boot interface.)  
 * DMI:  E3 Pro gaming v5
 * CPU:  xeon e3-1770 v6

RR version: (You can find it in the update menu.)  
 * RR:  24.12.4

DSM: 
 * model:  DS3622xs+
 * version:  7.2.2-72806-0

Issue:  
Is there a driver that can work with Adaptec 5405Z?
I was hoping to use this device but it didn't work.

My bild config

{"format":"img","language":"ru_RU","model":"DS3622xs+","version":"7.2.2-72806-0","addons":"acpid,mountloader,powersched,reboottoloader,trivial,cpuinfo,beep,codecpatch,expands,lsiutil,nvmesystem,remotefs,sensors,smartctl,updatenotify,wireless","modules":"8390,8139cp,8139too,9p,9pnet,9pnet_virtio,aacraid,acpi-cpufreq,acpi_call,adm1021,adm1031,adm9240,adt7470,adt7475,aic94xx,alx,apex,aqc111,asix,atkbd,atl1c,atl1e,atlantic,auxiliary,ax88179_178a,b44,bcm-phy-lib,be2net,bitblit,bnx2,bnx2x,bnx2x_mod,bnxt_en,broadcom,btcoexist,button,cdc_ether,cdc_ncm,cdrom,cfbcopyarea,cfbfillrect,cfbimgblt,cfg80211,ch,ch341,cls_fw,cls_u32,cn,cp210x,cpufreq_conservative,cpufreq_governor,cpufreq_ondemand,cpufreq_performance,cpufreq_powersave,cpufreq_stats,cpufreq_userspace,crc-ccitt,crc-itu-t,cxgb,cxgb3,cxgb4,cxgb4vf,dm9601,dme1737,dvb-core,dvb-usb-dib0700,dvb-usb-dvbsky,dvb-usb,dvb_usb_v2,e1000,e1000e,eeprom_93cx6,efivarfs,etxhci-hcd,evbug,evdev,exfat,f71882fg,f75375s,fb,fb_sys_fops,fbcon,fbdev,font,ftdi_sio,gasket,gpio-beeper,hpilo,hpsa,hpwdt,hvc_console,i2c-algo-bit,i40e,i8042,iavf,ib_addr,ib_cm,ib_core,ib_mad,ib_sa,ib_umad,ib_uverbs,ice,ide-core,ide-gd_mod,ide-pci-generic,igb,igbvf,igc,intel_auxiliary,intel_idle,ipv6,irqbypass,iscsi_tcp,it87,it8712f_wdt,it87_wdt,iw_cm,iwldvm,iwlmvm,iwlwifi,ixgb,ixgbe,ixgbevf,jc42,jme,kvm-amd,kvm-intel,kvm,lib80211,libfc,libiscsi,libiscsi_tcp,libphy,llc,lm75,lm78,lm90,lm95245,m88ds3103,mac80211,mdio,megaraid,megaraid_mbox,megaraid_mm,megaraid_sas,mii,mlx4_core,mlx4_en,mlx4_ib,mlx5_core,mlx5_ib,mlx_compat,mlxsw_core,mlxsw_pci,mmc_block,mmc_core,mpt3sas,mptbase,mptctl,mptsas,mptscsih,mptspi,mt7601u,mtk-sd,mvsas,mwifiex,mwifiex_sdio,mwifiex_usb,nct6683,nct6775,ne2k-pci,netxen_nic,ntfs,p8022,pcspeaker,pcspkr,phy-mt65xx-usb3,phy,piix,pktcdvd,pl2303,pmbus,pmbus_core,prism2_usb,processor,psnap,qcaspi,qed,qede,qedr,qla3xxx,qlcnic,qlge,r8125,r8125_txrx,r8126,r8126_txrx,r8152,r8168,r8168_tx,r8169,r8169_lk,rc-core,realtek,regmap-i2c,rndis_host,rtl8187,rtl8188ee,rtl818x_pci,rtl8192c-common,rtl8192ce,rtl8192cu,rtl8192de,rtl8192ee,rtl8192se,rtl8723-common,rtl8723ae,rtl8723be,rtl8821ae,rtl8xxxu,rtl_pci,rtl_usb,rtlwifi,rtsx_pci,rtsx_pci_sdmmc,rtsx_usb,rtsx_usb_sdmmc,sch_htb,sch_netem,sch_sfq,scsi_transport_fc,scsi_transport_spi,sdhci-acpi,sdhci-pci-data,sdhci-pci,sdhci-pltfm,sdhci,sdio_uart,sfc,sg,si2157,si2168,skge,sky2,softcursor,softdog,sp2,sr_mod,ssb,staging,stp,synobios,syscopyarea,sysfillrect,sysimgblt,tehuti,tg3,thermal,tps65217,ts2020,tty0tty,uhci-hcd,usbnet,usbserial,ushc,vfio,via-rhine,via-sdmmc,via-velocity,virtio,virtio_blk,virtio_console,virtio_input,virtio_mmio,virtio_net,virtio_pci,virtio_ring,virtio_scsi,vmw_pvscsi,vmw_vmci,vmxnet3,vub300,w83781d,w83793,watchdog,wch,wch_pre,wireguard"}

wjz304, 2024-12-12:
plz get the  vid and pid

eridanletalis, 2024-12-12:
> plz get the vid and pid

9005:0285 9005:02d8
PCI ID  9005:0285:9005:02d8 Adaptec 5405Z

wjz304, 2024-12-12:
Currently, aacraid cannot support

eridanletalis, 2024-12-12:
Are there any plans to add?

wjz304, 2024-12-12:
I compiled the driver and it doesn't work,
I don't have this hardware, I don't know what the error is, so I don't know how to fix it.

closed 2024-12-13 by eridanletalis