RROrg/rr #4611: BUG 引导不能从闪存拷贝内核模块

bsdcpp, 2024-12-13:
请填写以下信息.  
Please fill in the following information.  

Install ENV: (You can find it in the boot interface.)  
 * DMI:  dxp 4800plus
 * CPU:  8505
 * NIC:  (pid & vid)

RR version: (You can find it in the update menu.)  
 * RR:  24.12.4
 * addons:  acpid, trivial, mountloader, powersched, reboottoloader, beep, sensors, codecpatch, cpuinfo, smartctl
 * modules:  全选
 * lkms:  prod

DSM: 
 * model:  SA6400
 * version:  7.2 72806

Issue:  dd 整盘重做了引导，hdd全部拔掉。按照上述选项编译的引导，插上hdd后启动，发现it87的内核模块无法拷贝到/lib/modules下，/usr/bin/rr-sensors.sh无法加载it87驱动。之前还好的，能正确识别it8613传感器。

logs:  dmesg里没看到相关报错

感谢！

(## 因为 log中存在 SN/MAC 等一些敏感信息, 当提供完整文件时请自行抹除他们, 当然你也可以发送到我的邮箱. ##)  
(## Because the log contains some sensitive information such as SN/MAC, please delete them when providing the complete file. Of course, you can also send it to my email. ##)    
...

(请先看一下#173、#175、#226 的内容)  
(Plz review the content of #173, #175, #226 first)  
...

(如果你只是说 XXX 不能用, 什么详细信息也不提供, 我也只能说感谢你的反馈.)  
(If you just say XXX doesn't work without providing any details, I can only say thank you for your feedback.)  
...

VictoriousRaptor, 2024-12-14:
反馈，同样的问题，SA6400 7.2.1 69057 选择了it87模块，但进入DSM后lsmod看不到it87

bsdcpp, 2024-12-14:
> 反馈，同样的问题，SA6400 7.2.1 69057 选择了it87模块，但进入DSM后lsmod看不到it87
默认不启用fancontrol脚本(/usr/bin/rr-sensors.sh)的话是不加载的，可以试试以下两条，看看it87可以加载否
modprobe  hwmon-vid
modprobe  it87
我现在是从引导分区里拷贝it87.ko出来，测试可以正确识别，但是不优雅，我记得前几天明明是正常识别的。

naoki66, 2024-12-15:
> > 反馈，同样的问题，SA6400 7.2.1 69057 选择了it87模块，但进入DSM后lsmod看不到it87
> 
> 默认不启用fancontrol脚本(/usr/bin/rr-sensors.sh)的话是不加载的，可以试试以下两条，看看it87可以加载否
> 
> ```
> modprobe  hwmon-vid
> modprobe  it87
> ```
> 
> 我现在是从引导分区里拷贝it87.ko出来，测试可以正确识别，但是不优雅，我记得前几天明明是正常识别的。

我modprobe  hwmon-vid和modprobe  it87后运行/usr/bin/rr-sensors.sh报错：
At least one referenced file is missing. Either some required kernel
modules haven't been loaded, or your configuration file is outdated.
In the latter case, you should run pwmconfig again.

单独运行root@DSM:~# sensors
it8613-isa-0290
Adapter: ISA adapter
in0:           1.21 V  (min =  +0.00 V, max =  +2.81 V)
in1:           1.13 V  (min =  +0.00 V, max =  +2.81 V)
in2:           2.04 V  (min =  +0.00 V, max =  +2.81 V)
in4:           2.04 V  (min =  +0.00 V, max =  +2.81 V)
in5:           2.09 V  (min =  +0.00 V, max =  +2.81 V)
3VSB:          2.86 V  (min =  +0.00 V, max =  +5.61 V)
Vbat:          3.06 V  
+3.3V:         3.32 V  
fan2:         964 RPM  (min =    0 RPM)
fan3:        2385 RPM  (min =    0 RPM)
fan4:           0 RPM  (min =    0 RPM)
fan5:           0 RPM  (min =    0 RPM)
temp1:        +38.0°C  (low  = -128.0°C, high = +127.0°C)  sensor = thermal diode
temp2:        +37.0°C  (low  = -128.0°C, high = +127.0°C)  sensor = thermal diode
temp3:        +49.0°C  (low  = -128.0°C, high = +127.0°C)  sensor = Intel PECI
intrusion0:  ALARM

acpitz-acpi-0
Adapter: ACPI interface
temp1:        +27.8°C  (crit = +110.0°C)

coretemp-isa-0000
Adapter: ISA adapter
Package id 0:  +58.0°C  (high = +105.0°C, crit = +105.0°C)
Core 0:        +58.0°C  (high = +105.0°C, crit = +105.0°C)
Core 1:        +58.0°C  (high = +105.0°C, crit = +105.0°C)
Core 2:        +58.0°C  (high = +105.0°C, crit = +105.0°C)
-----------
按提示运行root@DSM:~# pwmconfig
-ash: pwmconfig: command not found

bsdcpp, 2024-12-15:
@naoki66 你这说明没有标题的bug存在，因为你insmod it87.ko正常的，你可以执行find /lib/modules -name 'it87*'看下应该是存在的。你现在都能识别it8613了，只不过你的/etc/fancontrol需要自己修改，因为它没有提供pwmconfig，你可以执行
```
ls /sys/class/hwmon/hwmon*/fan*_input
```
去看下hwmon后面的数字是几，几号风扇，自己对应修改好，然后运行rr-sensors.sh就会自动调节了

naoki66, 2024-12-15:
> @naoki66 你这说明没有标题的bug存在，因为你insmod it87.ko正常的，你可以执行find /lib/modules -name 'it87*'看下应该是存在的。你现在都能识别it8613了，只不过你的/etc/fancontrol需要自己修改，因为它没有提供pwmconfig，你可以执行
> 
> ```
> ls /sys/class/hwmon/hwmon*/fan*_input
> ```
> 
> 去看下hwmon后面的数字是几，几号风扇，自己对应修改好，然后运行rr-sensors.sh就会自动调节了

但是需要modprobe  才能识别哦，特意去看了下 /lib/modules 下存在这两个驱动，但是似乎没能自动载入。
不执行modprobe  只能看到
acpitz-acpi-0
Adapter: ACPI interface
temp1: +27.8°C (crit = +110.0°C)

coretemp-isa-0000
Adapter: ISA adapter
Package id 0: +58.0°C (high = +105.0°C, crit = +105.0°C)
Core 0: +58.0°C (high = +105.0°C, crit = +105.0°C)
Core 1: +58.0°C (high = +105.0°C, crit = +105.0°C)
Core 2: +58.0°C (high = +105.0°C, crit = +105.0°C)

bsdcpp, 2024-12-15:
@naoki66 正常，你可以去定时任务里启用那个fancontrol，默认RR都做好了，你用到了才会加载。

VictoriousRaptor, 2024-12-16:
> > 反馈，同样的问题，SA6400 7.2.1 69057 选择了it87模块，但进入DSM后lsmod看不到it87
> 
> 默认不启用fancontrol脚本(/usr/bin/rr-sensors.sh)的话是不加载的，可以试试以下两条，看看it87可以加载否
> 
> ```
> modprobe  hwmon-vid
> modprobe  it87
> ```
> 
> 我现在是从引导分区里拷贝it87.ko出来，测试可以正确识别，但是不优雅，我记得前几天明明是正常识别的。

认真看了之后，我的情况可能不太一样，`sensors-detect`提示找不到传感器。pe进了aida64和hwinfo也没有看到风扇。机型是HP 800 G3 SFF。

```
Some Super I/O chips contain embedded sensors. We have to write to
standard I/O ports to probe them. This is usually safe.
Do you want to scan for Super I/O sensors? (YES/no): y
Probing for Super-I/O at 0x2e/0x2f
Trying family `National Semiconductor/ITE'...               Yes
Found unknown chip with ID 0x1e11
Probing for Super-I/O at 0x4e/0x4f

```
root@NAS:~# modprobe it87
modprobe: ERROR: could not insert 'it87': No such device
```

bsdcpp, 2024-12-16:
@VictoriousRaptor  cpu型号呢，是否对应的是it87系支持的列表内传感器。

VictoriousRaptor, 2024-12-16:
> @VictoriousRaptor cpu型号呢，是否对应的是it87系支持的列表内传感器。

CPU G4560T，能获取到CPU温度。

```
Now follows a summary of the probes I have just done.
Just press ENTER to continue:
Driver `coretemp':
  * Chip `Intel digital thermal sensor' (confidence: 9)
```

bsdcpp, 2024-12-16:
看起来没检测到传感器，你看看bios里是啥，我主板里就是it8613
[image]
@VictoriousRaptor

wjz304, 2024-12-16:
Found unknown chip with ID 0x1e11

VictoriousRaptor, 2024-12-16:
> 看起来没检测到传感器，你看看bios里是啥，我主板里就是it8613 ![WechatIMG2350](https://private-user-images.githubusercontent.com/1034557/396130942-47edaa8f-0f36-48d2-ad10-f4e8eba93187.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzQzNTkyOTEsIm5iZiI6MTczNDM1ODk5MSwicGF0aCI6Ii8xMDM0NTU3LzM5NjEzMDk0Mi00N2VkYWE4Zi0wZjM2LTQ4ZDItYWQxMC1mNGU4ZWJhOTMxODcuanBnP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MTIxNiUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDEyMTZUMTQyMzExWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9OTY2MzZlMTUwNTdjZGViYmQ3N2JlZTRhMDA2MGY5YWI0ZmNiYjU4OThiYWQ2ZmQyYzQ1OGI1NGZiZjk2MjUyYSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.uZit3bjyAfT7s5CKGtsujeJheZc69396AN-8uH1iuzY) @VictoriousRaptor

HP品牌的主板，BIOS里面没有这些信息，估计是真没有传感器驱动了，windows pe 下面也看不到

wjz304, 2024-12-18:
modprobe -r it87
modprobe it87 force_id=0x1e11

试试吧

VictoriousRaptor, 2024-12-19:
> modprobe -r it87 modprobe it87 force_id=0x1e11
> 
> 试试吧

```
modprobe: ERROR: could not insert 'it87': No such device
```

wjz304, 2024-12-19:
反正我全网没搜到 0x1e11 是什么设备

closed 2025-02-17 by wjz304