RROrg/rr #4646: Acpid Tainted P, can not shutdown dsm7.0.1-42218 in pve8

suwey, 2024-12-19:
Thanks for saved my data first, but shutdown not working now because acpid could not start, also tried a patch that worked fine before. Pve verion 8.3.2, linux kernel 6.8.12-5-pve, DSM settings: q35 virt-gpu virt-scsi virt-io, here is the log from dbgutils:

`starting serial terminal on interface serial0
[   66.678473] BUG: unable to handle kernel paging request at 0000000067642b04
[   66.679377] IP: [<ffffffffa0652e69>] evdev_open+0x39/0x200 [evdev]
[   66.679377] PGD 0 
[   66.682504] Oops: 0000 [#8] SMP 
[   66.682504] Modules linked in: nfnetlink xfrm_user xfrm_algo fuse xt_ipvs ip_vs_rr ip_vs xt_mark iptable_mangle br_netfilter bridge stp aufs macvlan veth xt_conntrack xt_addrtype nf_conntrack_ipv6 nf_defrag_ipv6 ip6table_filter ip6_tables ipt_MASQUERADE xt_REDIRECT nf_nat_masquerade_ipv4 xt_nat iptable_nat nf_nat_ipv4 nf_nat_redirect nf_nat xt_recent xt_iprange xt_limit xt_state xt_tcpudp xt_multiport xt_LOG nf_conntrack_ipv4 nf_defrag_ipv4 nf_conntrack iptable_filter ip_tables x_tables 8021q vhost_scsi(O) vhost(O) tcm_loop(O) iscsi_target_mod(O) target_core_user(O) target_core_ep(O) target_core_multi_file(O) target_core_file(O) target_core_iblock(O) target_core_mod(O) syno_extent_pool(PO) rodsp_ep(O) udf isofs loop synoacl_vfs(PO) btrfs ecryptfs zstd_decompress zstd_compress xxhash xor raid6_pq aesni_intel glue_helper lrw gf128mul ablk_helper zram apollolake_synobios(PO) hid_generic usblp i915 drm_kms_helper syscopyarea sysfillrect sysimgblt fb_sys_fops cfbfillrect cfbcopyarea cfbimgblt drm drm_panel_orientation_quirks iosf_mbi fb fbdev video backlight i2c_algo_bit usb_storage sg dm_snapshot dm_bufio crc_itu_t crc_ccitt psnap p8022 llc hfsplus md4 hmac sit tunnel4 ipv6 flashcache_syno(O) flashcache(O) syno_flashcache_control(O) dm_mod arc4 crc32c_intel cryptd sha256_generic ecb aes_x86_64 authenc des_generic ansi_cprng cts md5 cbc cpufreq_powersave cpufreq_performance cpufreq_stats vxlan ip6_udp_tunnel udp_tunnel ip_tunnel kvm_amd(OE) kvm vfat fat etxhci_hcd pcspkr(OE) pcspeaker(OE) usbhid hid virtio_console(OE) virtio_net(OE) hvc_console(OE) atkbd(OE) evdev(OE) sr_mod(OE) irqbypass cdrom(OE) virtio_pci(OE) virtio(OE) virtio_ring(OE) rtc_cmos(OE) i8042(OE) processor button xhci_pci xhci_hcd uhci_hcd ehci_pci ehci_hcd usbcore usb_common [last unloaded: apollolake_synobios]
[   66.682504] CPU: 0 PID: 16133 Comm: acpid Tainted: P      D W  OE   4.4.180+ #42218
[   66.682504] Hardware name: QEMU Standard PC (Q35 + ICH9, 2009), BIOS rel-1.16.3-0-ga6ed6b701f0a-prebuilt.qemu.org 04/01/2014
[   66.682504] task: ffff8802ee0dcc80 ti: ffff8802ec428000 task.ti: ffff8802ec428000
[   66.682504] RIP: 0010:[<ffffffffa0652e69>]  [<ffffffffa0652e69>] evdev_open+0x39/0x200 [evdev]
[   66.682504] RSP: 0018:ffff8802ec42bc08  EFLAGS: 00010206
[   66.682504] RAX: ffffffffa0652e30 RBX: 0000000067642ae4 RCX: 00000000ffffffff
[   66.682504] RDX: 0000000000000040 RSI: 00000000024082c0 RDI: ffff8802e6fe0e80
[   66.682504] RBP: ffff8802ec42bc48 R08: 0000000000000000 R09: 0000000000000000
[   66.682504] R10: 0000000000000006 R11: 0000000000000000 R12: ffff8802e6fe0e80
[   66.682504] R13: ffff8802ec474500 R14: 0000000067642e44 R15: ffff8802ec474510
[   66.682504] FS:  00007fad240ef540(0000) GS:ffff8802fdc00000(0000) knlGS:0000000000000000
[   66.682504] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[   66.682504] CR2: 0000000067642b04 CR3: 00000002e41f3000 CR4: 00000000003406b0
[   66.682504] Stack:
[   66.682504]  0000000000000000 ffff8802ec474500 0001000000000001 ffff88007aaffb60
[   66.682504]  ffff8802e6fe0e80 ffff8802ec474500 ffffffffa06541e0 ffff8802ec474510
[   66.682504]  ffff8802ec42bc80 ffffffff81193fa0 ffff8802ec42bc80 ffff8802ec474500
[   66.682504] Call Trace:
[   66.682504]  [<ffffffff81193fa0>] chrdev_open+0x90/0x190
[   66.682504]  [<ffffffff81193f10>] ? cdev_put.part.0+0x20/0x20
[   66.682504]  [<ffffffff8118a842>] do_dentry_open+0x1e2/0x340
[   66.682504]  [<ffffffff8119b590>] ? __inode_permission+0x40/0xb0
[   66.682504]  [<ffffffff8118bc51>] vfs_open+0x51/0x80
[   66.682504]  [<ffffffff811a0765>] path_openat+0x595/0x1a40
[   66.682504]  [<ffffffff811999e2>] ? path_get+0x22/0x30
[   66.682504]  [<ffffffff811ab240>] ? dput.part.25+0x20/0x200
[   66.682504]  [<ffffffff811a390e>] do_filp_open+0x7e/0xc0
[   66.682504]  [<ffffffff8116041c>] ? handle_mm_fault+0x13bc/0x1660
[   66.682504]  [<ffffffff811b1c6b>] ? __alloc_fd+0x3b/0x170
[   66.682504]  [<ffffffff81189b4f>] do_sys_open+0x1af/0x240
[   66.682504]  [<ffffffff8118bf3f>] SyS_openat+0xf/0x20
[   66.682504]  [<ffffffff81579c4a>] entry_SYSCALL_64_fastpath+0x1e/0x8e
[   66.682504] Code: 41 57 41 56 41 55 41 54 4c 8d 27 53 48 8d 64 24 e8 4c 8b b7 60 02 00 00 4c 89 f3 48 89 75 c8 48 81 eb 60 03 00 00 be c0 82 40 02 <49> 8b 86 c0 fc ff ff 8b 80 d0 00 00 00 c1 e0 03 83 f8 40 48 0f 
[   66.758816] RIP  [<ffffffffa0652e69>] evdev_open+0x39/0x200 [evdev]
[   66.763970]  RSP <ffff8802ec42bc08>
[   66.763970] CR2: 0000000067642b04
[   66.767555] ---[ end trace cfba7f263d47c667 ]---
`

wjz304, 2024-12-19:
How do you shutdown? Do you click Shutdown in the DSM or in PVE ?

wjz304, 2024-12-19:
share /etc/pve/qemu-server/???.conf in pve

suwey, 2024-12-20:
@wjz304 upgrade dsm to 7.1.1 and install Qemu-ga then acpid works fine, thanks.

closed 2024-12-21 by wjz304