RROrg/rr #4670: BUG 群晖自带虚拟机Vittual Machie Manager不能创建任何虚拟设备

Liuruicheng520, 2024-12-22:
请填写以下信息.  
Please fill in the following information.  

Install ENV: (You can find it in the boot interface.)  
 * DMI:  
 * CPU:  i9 19900t
 * NIC:  (pid & vid)

RR version: (You can find it in the update menu.)  
 * RR:  24.12.4
 * addons:  
 * modules:  
 * lkms:  

DSM: 
 * model:  sa6400
 * version:  7.2.2-72806 update 2

群晖内置虚拟机 Vittual Machie Manager 不能创建任何虚拟设备 导入以前备份的虚拟设备 卡35%  重新创建新iso 不管windows系统还是软路由 op还是爱快  都不能开机  然后从群里面了解到 不是我个人出现这个问题 希望能解决这个问题
[image]
[image]
[image]

sd19870720, 2024-12-24:
是的，我也是一样的问题我的sa6400最新版DSM 7.2.2-72806 Update 2实体机和ds918虚拟机Intel Core i7-9700K版本是DSM 7.2.2-72806 Update 2都是这样

AisukaYuki, 2024-12-26:
我的一开始也是一直准备中，一直卡着什么都做不了，然后卸载重装vmm，重新创建了虚拟机不能开机，然后创建的虚拟机删不掉，导致集群创建的空间全都删不掉。最后进ssh把vmm相关的东西全删了。。。

Liuruicheng520, 2024-12-26:
> 我的一开始也是一直准备中，一直卡着什么都做不了，然后卸载重装vmm，重新创建了虚拟机不能开机，然后创建的虚拟机删不掉，导致集群创建的空间全都删不掉。最后进ssh把vmm相关的东西全删了。。。

具体怎么操作 是进入群晖的根目录 还是rr引导的目录，求个教程 感谢🙏

AisukaYuki, 2024-12-26:
卸载VMM套件,然后删除以下，
/volume1/@iSCSI/LUN/VDISK_BLUN  这里是创建的虚拟机
 /var/packages @appconf @appdata @apphome @appshare @apptemp 里面的virtualization是相关的配置
我不确定全删干净了没。反正用不了，都删了省的占空间。

Liuruicheng520, 2024-12-26:
> 卸载VMM套件,然后删除以下，
> 
> /volume1/@iSCSI/LUN/VDISK_BLUN  这里是创建的虚拟机
> 
>  /var/packages @appconf @appdata @apphome @appshare @apptemp 里面的virtualization是相关的配置
> 
> 我不确定全删干净了没。反正用不了，都删了省的占空间。

好的 我等会去尝试一下

Liuruicheng520, 2024-12-26:
> 卸载VMM套件,然后删除以下，
> 
> /volume1/@iSCSI/LUN/VDISK_BLUN  这里是创建的虚拟机
> 
>  /var/packages @appconf @appdata @apphome @appshare @apptemp 里面的virtualization是相关的配置
> 
> 我不确定全删干净了没。反正用不了，都删了省的占空间。

那现在你可以用群辉自带的虚拟机了吗？

AisukaYuki, 2024-12-26:
> > 卸载VMM套件,然后删除以下，
> > /volume1/@iSCSI/LUN/VDISK_BLUN  这里是创建的虚拟机
> > /var/packages @appconf @appdata @apphome @AppShare @apptemp 里面的virtualization是相关的配置
> > 我不确定全删干净了没。反正用不了，都删了省的占空间。
> 
> 那现在你可以用群辉自带的虚拟机了吗？

并不能 :<

Liuruicheng520, 2024-12-26:
> > > 卸载VMM套件,然后删除以下，
> 
> > > /volume1/@iSCSI/LUN/VDISK_BLUN  这里是创建的虚拟机
> 
> > > /var/packages @appconf @appdata @apphome @AppShare @apptemp 里面的virtualization是相关的配置
> 
> > > 我不确定全删干净了没。反正用不了，都删了省的占空间。
> 
> > 
> 
> > 那现在你可以用群辉自带的虚拟机了吗？
> 
> 
> 
> 并不能 :<

你的配置也是九代cpu吗？还有你也是切换的custom内核吗？

AisukaYuki, 2024-12-26:
> > > > 卸载VMM套件,然后删除以下，
> > 
> > 
> > > > /volume1/@iSCSI/LUN/VDISK_BLUN  这里是创建的虚拟机
> > 
> > 
> > > > /var/packages @appconf @appdata @apphome @AppShare @apptemp 里面的virtualization是相关的配置
> > 
> > 
> > > > 我不确定全删干净了没。反正用不了，都删了省的占空间。
> > 
> > 
> > > 
> > 
> > 
> > > 那现在你可以用群辉自带的虚拟机了吗？
> > 
> > 
> > 并不能 :<
> 
> 你的配置也是九代cpu吗？还有你也是切换的custom内核吗？

n5105 10代 Jasper Lake custom内核

Liuruicheng520, 2024-12-26:
> > > > > 卸载VMM套件,然后删除以下，
> 
> > > 
> 
> > > 
> 
> > > > > /volume1/@iSCSI/LUN/VDISK_BLUN  这里是创建的虚拟机
> 
> > > 
> 
> > > 
> 
> > > > > /var/packages @appconf @appdata @apphome @AppShare @apptemp 里面的virtualization是相关的配置
> 
> > > 
> 
> > > 
> 
> > > > > 我不确定全删干净了没。反正用不了，都删了省的占空间。
> 
> > > 
> 
> > > 
> 
> > > > 
> 
> > > 
> 
> > > 
> 
> > > > 那现在你可以用群辉自带的虚拟机了吗？
> 
> > > 
> 
> > > 
> 
> > > 并不能 :<
> 
> > 
> 
> > 你的配置也是九代cpu吗？还有你也是切换的custom内核吗？
> 
> 
> 
> n5105 10代 Jasper Lake custom内核

是不是用official内核启动不了 我目前只能custom启动

AisukaYuki, 2024-12-26:
> 是不是用official内核启动不了 我目前只能custom启动

你要说群晖引导启动的话，我两个内核都没问题。

Liuruicheng520, 2024-12-26:
> > > > > > > 卸载VMM套件,然后删除以下，
> 
> > > 
> 
> > > 
> 
> > > > >
> 
> > > 
> 
> > > 
> 
> > > > >
> 
> > > 
> 
> > > 
> 
> > > > > > > /volume1/@iSCSI/LUN/VDISK_BLUN  这里是创建的虚拟机
> 
> > > 
> 
> > > 
> 
> > > > >
> 
> > > 
> 
> > > 
> 
> > > > >
> 
> > > 
> 
> > > 
> 
> > > > > > > /var/packages @appconf @appdata @apphome @AppShare @apptemp 里面的virtualization是相关的配置
> 
> > > 
> 
> > > 
> 
> > > > >
> 
> > > 
> 
> > > 
> 
> > > > >
> 
> > > 
> 
> > > 
> 
> > > > > > > 我不确定全删干净了没。反正用不了，都删了省的占空间。
> 
> > > 
> 
> > > 
> 
> > > > >
> 
> > > 
> 
> > > 
> 
> > > > >
> 
> > > 
> 
> > > 
> 
> > > > > >
> 
> > > 
> 
> > > 
> 
> > > > >
> 
> > > 
> 
> > > 
> 
> > > > >
> 
> > > 
> 
> > > 
> 
> > > > > > 那现在你可以用群辉自带的虚拟机了吗？
> 
> > > 
> 
> > > 
> 
> > > > >
> 
> > > 
> 
> > > 
> 
> > > > >
> 
> > > 
> 
> > > 
> 
> > > > > 并不能 :<
> 
> > > 
> 
> > > 
> 
> > > > 
> 
> > > 
> 
> > > 
> 
> > > > 你的配置也是九代cpu吗？还有你也是切换的custom内核吗？
> 
> > > 
> 
> > > 
> 
> > > n5105 10代 Jasper Lake custom内核
> 
> > 
> 
> > 是不是用official内核启动不了 我目前只能custom启动
> 
> 
> 
> 你要说群晖引导启动的话，我两个内核都没问题。但是vmm也都用不了。

我看别人13代 office内核 sa6400可以启动 且可以启用群晖 虚拟机[image]

Liuruicheng520, 2024-12-26:
> 你要说群晖引导启动的话，我两个内核都没问题。但是vmm也都用不了。

[image]

Liuruicheng520, 2024-12-26:
> 你要说群晖引导启动的话，我两个内核都没问题。但是vmm也都用不了。

这是昨天TG群 找的一个哥们 给我测试刚安装的 小白鼠机器 然而他居然可以启动 我瞬间石化 ~~~

sd19870720, 2024-12-29:
内核换了也不行，删的干干净净再安再搞都是不行的

Liuruicheng520, 2024-12-29:
> 内核换了也不行，删的干干净净再安再搞都是不行的

刚更新最新版本 还是凉凉 要不试试降级？群晖系统

sd19870720, 2024-12-30:
> > 内核换了也不行，删的干干净净再安再搞都是不行的
> 
> 刚更新最新版本 还是凉凉 要不试试降级？群晖系统

降级后总会出些莫名奇妙的小毛病，要不再等等大神们修复这个大毛病吧

Liuruicheng520, 2024-12-30:
> > > 内核换了也不行，删的干干净净再安再搞都是不行的
> 
> > 
> 
> > 刚更新最新版本 还是凉凉 要不试试降级？群晖系统
> 
> 
> 
> 降级后总会出些莫名奇妙的小毛病，要不再等等大神们修复这个大毛病吧

我切换内核直接启动不了 屏幕也不会熄屏 但是可以看到ip 自带的虚拟机不能启动也不知道会不会解决这个问题

wjz304, 2024-12-30:
> > > 内核换了也不行，删的干干净净再安再搞都是不行的
> > 
> > 
> > 刚更新最新版本 还是凉凉 要不试试降级？群晖系统
> 
> 降级后总会出些莫名奇妙的小毛病，要不再等等大神们修复这个大毛病吧

别等了，
假如 你炒了个西红柿鸡蛋，别人都说好吃，我说不好吃，期待明天的好吃！
你明天就知道咋炒好吃了？

Liuruicheng520, 2024-12-31:
试试16版本 看有没有修复什么问题

sd19870720, 2024-12-31:
> 试试16版本 看有没有修复什么问题

刚升级了试了不行

Liuruicheng520, 2024-12-31:
> > 试试16版本 看有没有修复什么问题
> 
> 刚升级了试了不行

实在不行降级吧 不能虚拟设备多浪费性能多核

sd19870720, 2024-12-31:
> > > 试试16版本 看有没有修复什么问题
> > 
> > 
> > 刚升级了试了不行
> 
> 实在不行降级吧 不能虚拟设备多浪费性能多核

没换机型降到7.1.1了也不行

Liuruicheng520, 2024-12-31:
> > > > 试试16版本 看有没有修复什么问题
> > > 
> > > 
> > > 刚升级了试了不行
> > 
> > 
> > 实在不行降级吧 不能虚拟设备多浪费性能多核
> 
> 没换机型降到7.1.1了也不行

隔壁网 不是有个留言23.版本的rr 可以虚拟机 感觉问题出在高版本的rr

IvanWoo22, 2025-01-02:
rr: 24.12.6
dsm: 72806
cpu: J4125
亲测三个虚拟机都可以正常运行

Liuruicheng520, 2025-01-02:
> rr: 24.12.6 dsm: 72806 cpu: J4125 亲测三个虚拟机都可以正常运行

问题应该出现在第9代  第10代 cpu这边 不是说所有都出问题

luyanfeng, 2025-01-02:
有用虚拟机需求的话, 还是用pve之类的 虚拟化系统吧

JiangGeek, 2025-01-02:
目前我几个虚拟机都正常，包括windows，Linux都正常，配置版本如下
rr:24.11.1
dms:72806
cpu:i7-8700k

Liuruicheng520, 2025-01-02:
> 目前我几个虚拟机都正常，包括windows，Linux都正常，配置版本如下 rr:24.11.1 dms:72806 cpu:i7-8700k

晚点试试你的24.11.1版本~~~

JiangGeek, 2025-01-02:
> > 目前我几个虚拟机都正常，包括windows，Linux都正常，配置版本如下 rr:24.11.1 dms:72806 cpu:i7-8700k
> 
> 晚点试试你的24.11.1版本~~~
补充一点是，内核选的official

Liuruicheng520, 2025-01-02:
> > > 目前我几个虚拟机都正常，包括windows，Linux都正常，配置版本如下 rr:24.11.1 dms:72806 cpu:i7-8700k
> > 
> > 
> > 晚点试试你的24.11.1版本~~~
> > 补充一点是，内核选的official

那你有条件 尝试换一下custom 看能不能启动 内部虚拟机  因为我们环境有限 我切换启动不了 nas

JiangGeek, 2025-01-09:
> > > > 目前我几个虚拟机都正常，包括windows，Linux都正常，配置版本如下 rr:24.11.1 dms:72806 cpu:i7-8700k
> > > 
> > > 
> > > 晚点试试你的24.11.1版本~~~
> > > 补充一点是，内核选的official
> 
> 那你有条件 尝试换一下custom 看能不能启动 内部虚拟机 因为我们环境有限 我切换启动不了 nas

昨天升级引导顺便也测试了下，切换到custom内核没办法启动VMM，official没问题

bao3, 2025-03-01:
12代 CPU，VMM 和大家症状一样，不用折腾了，应该是真的不行。

guanglin1994, 2025-06-03:
n6005  11代 选的custom 也是这个症状   这个问题有没有最新消息哇

zhoyed, 2025-08-09:
我的也是一样，SA6400在custom内核下虚拟机无法启动，official内核一切正常，我的处理器是双路E5 2630

guanglin1994, 2025-08-09:
这是来自QQ邮箱的假期自动回复邮件。
 
您好，我最近正在休假中，无法亲自回复您的邮件。我将在假期结束后，尽快给您回复。