RROrg/rr #4775: BUG : After 24.12.5 update, the SATA SSD disk is missing from the storage pool.

azchoi, 2024-12-29:
请填写以下信息.  
Please fill in the following information.  

Install ENV: (You can find it in the boot interface.)  
 * DMI:  
 * CPU:  Intel Xeon
 * NIC:  (pid & vid) Intel 10G NIC

RR version: (You can find it in the update menu.)  
 * RR:  24.12.5
 * addons:  N/A
 * modules:  N/A
 * lkms:  N/A

DSM: 
 * model:  3622XS
 * version:  7.2.2-72806-U2

Issue:  

**1. After 24.12.5 update, the SATA SSD disk is missing from the storage pool.
2. The missing SSD is a Samsung 860 PRO, and other SATA HDDs work fine.
3. The SSD that disappeared when downgraded to 24.12.4 appeared normally in the storage pool again.
4. For the above reasons, I think it is an issue for this loader.
---------中文翻译器------------
1. 24.12.5 更新后，存储池中缺少 SATA SSD 磁盘。
2. 丢失的SSD是Samsung 860 PRO，其他SATA硬盘正常工作。
3. 降级到24.12.4时消失的SSD再次正常出现在存储池中。
4. 出于上述原因，我认为这是这个加载器的问题**。

logs:  

(## 因为 log中存在 SN/MAC 等一些敏感信息, 当提供完整文件时请自行抹除他们, 当然你也可以发送到我的邮箱. ##)  
(## Because the log contains some sensitive information such as SN/MAC, please delete them when providing the complete file. Of course, you can also send it to my email. ##)    
...

(请先看一下#173、#175、#226 的内容)  
(Plz review the content of #173, #175, #226 first)  
...

(如果你只是说 XXX 不能用, 什么详细信息也不提供, 我也只能说感谢你的反馈.)  
(If you just say XXX doesn't work without providing any details, I can only say thank you for your feedback.)  
...

wjz304, 2024-12-29:
```
cat /sys/block/sd*/uevent
grep portcfg /etc/synoinfo.conf
synodiskport -internal
synodiskport -usb
```

wjz304, 2024-12-29:
test this: 
https://github.com/RROrg/rr-build/actions/runs/12535559619

Kaelber, 2024-12-29:
I confirm, that after update the storage are missing and have a critical situation. As I´m test only in a VM from a OVA is no problem to create a new machine, but the situation is the same. The predefined storage 10+32 GB are not usable in DSM. If I create a storage no. 3, after a storage apear in DSM as storage no. 2 and I can use.

wjz304, 2024-12-29:
https://github.com/RROrg/rr/releases/tag/24.12.6

azchoi, 2024-12-30:
> https://github.com/RROrg/rr/releases/tag/24.12.6

In case of 24.12.6, the issue have been cleared. Thank you so much.

closed 2024-12-30 by wjz304