RROrg/rr #4947: 升级总是失败

luyanfeng, 2024-12-31:
更新rr
RR v24.8.2 DS920+(geminilake） 7.2(72806） 2030SBRHR5XTM 192.168.0.107（qwerty/us） [BI0S]

提示 : 
Error extracting update file.

我手动上传updateall-xx.zip文件也是一样的错

luyanfeng, 2024-12-31:
原系统:
RR v24.8.2 DS920+(geminilake） 7.2(72806） 2030SBRHR5XTM 192.168.0.107（qwerty/us） [BI0S]

升级rr 到 RR v24.12.6

wjz304, 2024-12-31:
https://rrorg.cn/archives/rrupdate

ctl26481, 2024-12-31:
内存不够(例如2GB内存)会造成/tmp下载updateall-XX.zip后不够放解压的update档
可以手动将updateall-XX.zip放入空间够大的p3分区
再從/tmp创建updateall-XX.zip，softlink指向/mnt/p3/updateall-XX.zip，这样/tmp就有空间做解压了

大略是这样的:(若updateall-XX.zip已存在于/tmp)
```
cd /tmp
mv updateall-XX.zip /mnt/p3
touch updateall-XX.zip #在/tmp创建空档案
ln -s /mnt/p3/updateall-XX.zip /tmp/updateall-XX.zip #使用softlink将p3分区的实体档案指向/tmp的空档案
```

重新执行loader update，若空间还是不够则要先将/tmp解压出来的档案删除
完成后可将/mnt/p3/updateall-XX.zip删除，避免佔用空间

MrYWH, 2025-01-01:
能用就不升级，升级12.05缺盘我还以为盘坏了折腾好几天

luyanfeng, 2025-01-02:
> 内存不够(例如2GB内存)会造成/tmp下载updateall-XX.zip后不够放解压的update档 可以手动将updateall-XX.zip放入空间够大的p3分区 再從/tmp创建updateall-XX.zip，softlink指向/mnt/p3/updateall-XX.zip，这样/tmp就有空间做解压了
> 
> 大略是这样的:(若updateall-XX.zip已存在于/tmp)
> 
> ```
> cd /tmp
> mv updateall-XX.zip /mnt/p3
> touch updateall-XX.zip #在/tmp创建空档案
> ln -s /mnt/p3/updateall-XX.zip /tmp/updateall-XX.zip #使用softlink将p3分区的实体档案指回/tmp的空档案
> ```
> 
> 重新执行loader update，若空间还是不够则要先将/tmp解压出来的档案删除 完成后可将/mnt/p3/updateall-XX.zip删除，避免佔用空间

感谢, 根据您的给的方案, 我给pve加大到4g内存后成功了.
我也只是测试不新的rr能不能实用于我的机器升级. 
目前主机还也没升及. 只是在一个虚拟机上模拟来着.

luyanfeng, 2025-01-02:
> 能用就不升级，升级12.05缺盘我还以为盘坏了折腾好几天

感谢, 
我也只是测试新的rr能不能直 接用于我的机器升级. 
目前主机还也没升级, 只是在一个虚拟机上模拟来着.

closed 2025-01-02 by luyanfeng