RROrg/rr #4956: 保留内存过大的问题怎么解决啊？

sclzyz, 2025-01-02:
之前7.2.1用的arpl的引导文件，开机也是保留内存过大，查了资料，在引导的配置文件中添加了参数disable_mtrr_trim就解决了这个问题，现在新升级7.2.2改用了您的rr引导，同样的问题依葫芦画瓢，修改了grub.cfg文件，也这样改了
`# 修改RR_CMDLINE定义，添加disable_mtrr_trim
set RR_CMDLINE="earlyprintk earlycon=uart8250,io,0x3f8,115200n8 console=ttyS0,115200n8 root=/dev/ram rootwait intremap=off net.ifnames=0 panic=5 split_lock_detect=off pcie_aspm=off intel_pstate=disable nox2apic nomodeset disable_mtrr_trim"`

` linux /zImage-dsm ${dsm_cmdline} disable_mtrr_trim  # 添加在此处`

但是重启后没有效果，请问这个怎么解决啊？

[image]

luyanfeng, 2025-01-02:
为什么我的这么少

[image]

sclzyz, 2025-01-02:
就是不清楚才来求助的嘛，哪位大神指教一下

wjz304, 2025-01-02:
[image]
依葫芦画瓢  为什么 arpl 在 cmdline里加，rr要在 grub.cfg 里加？

closed 2025-01-03 by wjz304

sclzyz, 2025-01-18:
@wjz304 完美解决，非常感谢

[image]