RROrg/rr #4957: cpu无法降频，始终运行在最大频率

JimiZhou, 2025-01-02:
请填写以下信息.  
Please fill in the following information.  

Install ENV: (You can find it in the boot interface.)  
 * DMI:  Asus W480 Pro ACE
 * CPU:  Intel W1290P ES
 * NIC:  (pid & vid)

RR version: (You can find it in the update menu.)  
 * RR:  [24.12.6]
 * addons:  
 * modules:  
[image]

 * lkms:  

DSM: 
 * model:  SA6400
 * version:  7.2.2-72806 Update 2

Issue:  

logs:  

(## 因为 log中存在 SN/MAC 等一些敏感信息, 当提供完整文件时请自行抹除他们, 当然你也可以发送到我的邮箱. ##)  
(## Because the log contains some sensitive information such as SN/MAC, please delete them when providing the complete file. Of course, you can also send it to my email. ##)    
...

(请先看一下#173、#175、#226 的内容)  
(Plz review the content of #173, #175, #226 first)  
...

(如果你只是说 XXX 不能用, 什么详细信息也不提供, 我也只能说感谢你的反馈.)  
(If you just say XXX doesn't work without providing any details, I can only say thank you for your feedback.)  
... 

[image]

JimiZhou, 2025-01-02:
CPU基频是3.7，睿频群晖不可用，但是实际s-tui上显示跑到了4.2Ghz左右。尝试过cpufreq-set -g powersave -c {0..19}把所有CPU都设置成powersave策略，cpufreq-info显示800Mhz但实际还是顶着高频率在跑，是否有办法支持频率调节？

VASH-123, 2025-01-03:
[image]
[image]
我的可以

watermankao, 2025-01-04:
可以测一下cpu功耗

ZhangTianyi-1998, 2025-03-21:
关注   我也存在这个问题

wjz304, 2025-04-28:
https://github.com/RROrg/rr/blob/main/guide.md#:~:text=%23%20CPU%0Acat%20/sys/devices,%E8%AE%BE%E7%BD%AE%20CPU%20%E8%AE%BE%E7%BD%AE%E9%A2%91%E7%8E%87

closed 2025-05-08 by wjz304