RROrg/rr #5243: 求教，增加PCIE SATA后无法在DSM中识别

IvanWoo22, 2025-01-11:
最近新加了一块PCIe to SATA的卡（asmedia `ASM1061`主控），并连接了新的硬盘。

目前可以看到硬盘通电，rr里能看到这个SATA信息（不过识别成了ASM1062）以及两个新增的ports（目前只接了一个）。

重新编译后进入系统，依然只有原本的两个硬盘。按照 #1046 尝试后没有解决（DSM内原先显示12有盘，改过后显示34有盘），DSM中没有任何新增硬盘的痕迹/提示。

Install ENV: (You can find it in the boot interface.)  
 * CPU:  J4125
 * 主板:  映泰J4125NHU

RR version: (You can find it in the update menu.)  
 * RR:  25.1.0
 * addons:  ASM1061

DSM: 
 * model:  918+
 * version:  7.2.2

wjz304, 2025-01-13:
取  pid bid

IvanWoo22, 2025-01-13:
> 取 pid vid

请问是要我提供引导U盘的pid和vid吗？

IvanWoo22, 2025-01-17:
[image]

[image]

[image]

wjz304, 2025-01-17:
ASM1062 Serial ATA Controller [1b21:0612]

IvanWoo22, 2025-01-17:
嗯嗯，那应该如何将这个PCIE添加到DSM呢？目前群晖里识别不到新接在ASM1062的硬盘。

VASH-123, 2025-01-18:
这个芯片的老固件有bug，要更新到最新4.27V2才能用。https://imnks.com/3535.html

IvanWoo22, 2025-01-18:
> 这个芯片的老固件有bug，要更新到最新4.27V2才能用。https://imnks.com/3535.html

谢谢建议，我试了把这个插到Windows上并按照这个链接里说的更新了固件，插回NAS还是不行😑

VASH-123, 2025-01-19:
> > 这个芯片的老固件有bug，要更新到最新4.27V2才能用。https://imnks.com/3535.html
> 
> 谢谢建议，我试了把这个插到Windows上并按照这个链接里说的更新了固件，插回NAS还是不行😑

我有一块这个芯片4个口的，以前在win10下很好用，但是群晖下只能识别2个口我就没用它了，换了9215挺好的。

IvanWoo22, 2025-01-20:
> > > 这个芯片的老固件有bug，要更新到最新4.27V2才能用。https://imnks.com/3535.html
> > 
> > 
> > 谢谢建议，我试了把这个插到Windows上并按照这个链接里说的更新了固件，插回NAS还是不行😑
> 
> 我有一块这个芯片4个口的，以前在win10下很好用，但是群晖下只能识别2个口我就没用它了，换了9215挺好的。

好像确实是这个芯片不太行，我换了一块硬盘插上，能看到新盘了但还是各种操作不了，没法抹盘，准备放弃了

closed 2025-02-07 by IvanWoo22