RROrg/rr #5251: BUG

jetloga, 2025-01-12:
请填写以下信息.  
Please fill in the following information.  

Install ENV: (You can find it in the boot interface.)  
 * DMI:  To Be Filled By O.E.M. To Be Filled By O.E.M./Z370M Pro4, BIOS P4.20G 04/01/2021
 * CPU:  Intel(R) Core(TM) i3-9100T CPU @ 3.10GHz
 * NIC:  (pid & vid) 15b3:1007 Ethernet controller: Mellanox Technologies MT27520 Family [ConnectX-3 Pro]

RR version: (You can find it in the update menu.)  
 * RR:  25.1.0
 * addons:  25.1.0
 * modules:  25.1.0
 * lkms:  25.1.0

DSM: 
 * model:  ds920+
 * version:  7.2(72806)

Issue:  
更新到最新rr版本后dsm内核panic，打印错误信息：
[    0.315489] Kernel panic - not syncing: BIOS has enabled x2apic but kernel doesn't support x2apic, please disable x2apic in BIOS.
但是bios菜单中没有x2apic相关选项，主板是asrock z370m pro4
logs:  
[image]

(## 因为 log中存在 SN/MAC 等一些敏感信息, 当提供完整文件时请自行抹除他们, 当然你也可以发送到我的邮箱. ##)  
(## Because the log contains some sensitive information such as SN/MAC, please delete them when providing the complete file. Of course, you can also send it to my email. ##)    
...

(请先看一下#173、#175、#226 的内容)  
(Plz review the content of #173, #175, #226 first)  
...

(如果你只是说 XXX 不能用, 什么详细信息也不提供, 我也只能说感谢你的反馈.)  
(If you just say XXX doesn't work without providing any details, I can only say thank you for your feedback.)  
...

wjz304, 2025-01-13:
一般是有的，或者 删除 grub.cfg 里的 nox2apic 试试吧， 提前说明 apollolake，geminilake，purley 是不支持 x2apic  的

wjz304, 2025-01-13:
改直接启动也可以

jetloga, 2025-01-13:
尝试了一下，启动命令中已经不包含nox2apic，但问题依旧
[    0.000000] Command line: intel_idle.max_cstate=1 withefi mitigations=off earlyprintk syno_hw_version=DS920+ i915.enable_guc=3 console=ttyS0,115200n8 consoleblank=600 netif_num=2 pid=0x5571 earlycon=uart8250,io,0x3f8,115200n8 mac2=xxx mac1=xxx sn=xxx modprobe.blacklist=evbug,cdc_ether,mpt3sas pcie_aspm=off rootwait vid=0x0781 elevator=elevator loglevel=15 HddHotplug=1 skip_vender_mac_interfaces=0,1,2,3,4,5,6,7 intel_iommu=igfx_off i915.disable_power_well=1 DiskIdxMap=00 panic=3 vender_format_version=2 log_buf_len=32M root=/dev/md0 SataPortMap=6 syno_ttyS1=serial,0x2f8 syno_ttyS0=serial,0x3f8,115200 mev=physical kexecboot

wjz304, 2025-01-13:
apollolake，geminilake，purley 是不支持 x2apic 的

closed 2025-02-17 by wjz304