RROrg/rr #5293: synology photo 中删除照片请求一直在处理中

qiezihwy, 2025-01-13:
Install ENV:  
 * DMI:  qemu standard pc (Q35+ICH9, 2009) BIOS 4.2023.08-4 02/15/2024
 * CPU:  Intel(R) Pentium(R) CPU G4600 @ 3.60GHz
 * NIC:  pid 0x0001 vid 0x46f4

RR version: (You can find it in the update menu.)  
 * RR:  25.1.0
 * addons:  updatenotify acpid mountloader hdddb powersched reboottolloader 及系统引导自带
 * modules:  
 * lkms:  

DSM: 
 * model:  DS918+
 * version:  DSM 7.2.2-72806 Update2 

Issue:  
 synology photo 中删除照片请求一直在处理中，查看@appdata 中synology photo的报错，看到如下日志
logs:  
synofoto.log >>> 如下
2025-01-13T17:43:33+08:00 DS synofoto-task-center[10418]: /source/synofoto/src/lib/io/channel.cpp:82 channel[/run/synofoto/js-server-backend.socket] construct failed: connect: No such file or directory [system:2 at /usr/local/x86_64-pc-linux-gnu/x86_64-pc-linux-gnu/sys-root/usr/include/boost/asio/detail/reactive_socket_service.hpp:587:5 in function 'boost::system::error_code boost::asio::detail::reactive_socket_service<Protocol>::connect(implementation_type&, const endpoint_type&, boost::system::error_code&)']
2025-01-13T17:43:33+08:00 DS synofoto-task-center[10418]: /source/synofoto/src/lib/websocket/socket_notification.cpp:21 send index task failed. io error

wulalaxing, 2025-01-22:
我也遇到了这个问题，希望大佬能给予解决方法

tsingter, 2025-02-10:
真是个头疼的问题

huahuahuahuahuahuahuahua, 2025-02-25:
同，实际上已经删了，只是web没收到删除完成的通知

MaoZ1993, 2025-03-09:
奇怪了，我也遇到了这个问题，到白裙218play上测试同一套件版本则没有这个bug

lianlianlianlianlianlian, 2025-03-15:
套件的问题 降级到这个版本即可
https://archive.synology.cn/download/Package/SynologyPhotos/1.7.0-0795

lalakiop, 2025-03-26:
我也是碰到同样的问题

cjwkof, 2025-04-02:
也有这个问题，我F12删除完刷新请求是到了一个无效的URL

cjwkof, 2025-04-02:
也有这个问题，我删除完，F12看刷新请求是到了一个无效的URL

cjwkof, 2025-04-07:
更新到1.8beta还是存在这个问题，我估计是黑裙才有的，不然这么明显的问题官方应该会解决了

wjz304, 2025-04-08:
/var/packages/SynologyPhotos/target/usr/bin/synofoto-bin-push-service --get
在非白码下获取失败，导致 pkg-SynologyPhotos-js-server.service 服务启动失败.

wulalaxing, 2025-04-08:
有什么解决方案吗

wjz304, 2025-04-09:
下个版本加个插件吧

wjz304, 2025-04-11:
https://github.com/RROrg/rr/releases/tag/25.4.0

lingshengjiang898, 2025-04-12:
安装25.4.0版本并且启用apppatch 插件后会导致群晖桌面无法进入（群晖Assistant能发现但是无法进入群晖桌面），卸载apppatch后回复正常

refined-fish, 2025-04-12:
更新了25.4，也添加了apppatch插件，编译启动之后删除照片还是加载中
检查了 `/var/packages/SynologyPhotos/target/usr/bin/` 路径，已存在 `synofoto-bin-push-service` 这个文件，cat也是有内容的😫

closed 2025-04-13 by wjz304

qiezihwy, 2025-05-20:
> Install ENV:
>
> * DMI:  qemu standard pc (Q35+ICH9, 2009) BIOS 4.2023.08-4 02/15/2024
> * CPU:  Intel(R) Pentium(R) CPU G4600 @ 3.60GHz
> * NIC:  pid 0x0001 vid 0x46f4
>
> RR version: (You can find it in the [...]
在最新的synology photos 更新中已经更新了这个问题，跟黑裙没有关系，感谢大家关注，感谢@RROrg 让我们在issue中讨论跟rr无关的问题