RROrg/rr #5387: BUG wlanX name cause coredump

Kali-, 2025-01-15:
请填写以下信息.  
Please fill in the following information.  

Install ENV: (You can find it in the boot interface.)  
 * DMI:  
 * CPU:  
 * NIC:  (pid & vid)

RR version: (You can find it in the update menu.)  
 * RR:  
 * addons:  
 * modules:  
 * lkms:  

DSM: 
 * model:  xs
 * version:  7.2.2

Issue: 
loading and configuring a wifi device named wlan0 or wlan1 (i didnt tryed more) cause issue. Network page in Info Center is completly empty and everytime the tab is selected cause a coredump (journalctl -f). To be picky, the issue do not start instantly after configuring the devices, but after few minutes, and will be broken until device is removed (rmmod) or taken down (ifconfig wlan0 down)
i have found a workaround, loading the wifi modules with a ovs_ prefix seems to prevent this issues. I have used ovs_ as i suppose the OpenVirtual Switch which are also used by DSM (guess it comes used when virtualization is used) are handled diffently

example:
insmod /lib/modules/8188eu.ko ifname=ovs_wlan0

logs:  

(## 因为 log中存在 SN/MAC 等一些敏感信息, 当提供完整文件时请自行抹除他们, 当然你也可以发送到我的邮箱. ##)  
(## Because the log contains some sensitive information such as SN/MAC, please delete them when providing the complete file. Of course, you can also send it to my email. ##)    
...

(请先看一下#173、#175、#226 的内容)  
(Plz review the content of #173, #175, #226 first)  
...

(如果你只是说 XXX 不能用, 什么详细信息也不提供, 我也只能说感谢你的反馈.)  
(If you just say XXX doesn't work without providing any details, I can only say thank you for your feedback.)  
...

Kali-, 2025-01-15:
nvm, ovs_wlan0 still cause issue, happened after 30 minutes!
ill continue to dig

Kali-, 2025-01-15:
issue still happen, ill dig more before open it

wjz304, 2025-01-16:
Try using eth99 ?

Kali-, 2025-01-16:
right now i'm using ovs_eth9, looks promising but last night i got a power failure and my nas rebooted
ill continue to monitor 
did u prefer eth99 or ovs_eth99 ?

wjz304, 2025-01-16:
> right now i'm using ovs_eth9, looks promising but last night i got a power failure and my nas rebooted ill continue to monitor did u prefer eth99 or ovs_eth99 ?

I just thought of 99 after seeing your explanation, but 99 seems to be reserved for the management port.

wjz304, 2025-01-16:
As a general script, we can add a base number, such as 20, 90, wlan0 -> ovs_eth90

wjz304, 2025-01-16:
[ -n "$(cat /usr/syno/etc/synoovs/ovs_reg.conf 2>/dev/null)" ] && echo "Used OVS"

Kali-, 2025-01-17:
he "workaround" of use ovs_eth90 still working after 1 day :)

wjz304, 2025-01-17:
ovs_wlan0  你试过吗？

Kali-, 2025-01-17:
ovs_wlan0 still crash, its the 1st test u did

wjz304, 2025-01-17:
OK

> ovs_wlan0仍然崩溃，这是我做的第一次测试

Kali-, 2025-01-17:
also ovs_wlan0 is not shown in Network pages

wjz304, 2025-01-17:
嗯，我现在没环境，稍后我测试下，优化下脚本

wjz304, 2025-01-18:
It seems that ovs* is not needed, just rename wlanX to ethX

Kali-, 2025-01-19:
renamed to eth90

wjz304, 2025-01-20:
中国新年即将到来，我回老家了，大概要休息7–14天，等回去再更新addon

wjz304, 2025-01-27:
https://github.com/RROrg/rr/releases/tag/25.1.3

Kali-, 2025-01-27:
i have issue after 25.1.4 update

after update i get 2 devices eth70 and ovs_eth70
if i run manually 
bash -x /usr/bin/wireless_supplicant.sh eth70 ssid pass
synonet return error

`+ /usr/syno/sbin/synonet --dhcp eth70
synonet.c:214 Error: Invalid network interface [eth70]. `

i have tried to use ovs_eth70 and things is slight better but i dont get the ip it stuck with the zeroconf ip (i think something fails at wpa_supplicant level)

wjz304, 2025-02-06:
> i have issue after 25.1.4 update
> 
> after update i get 2 devices eth70 and ovs_eth70 if i run manually bash -x /usr/bin/wireless_supplicant.sh eth70 ssid pass synonet return error
> 
> ```
> + /usr/syno/sbin/synonet --dhcp eth70
> synonet.c:214 Error: Invalid network interface [eth70]. 
> ```
> 
> i have tried to use ovs_eth70 and things is slight better but i dont get the ip it stuck with the zeroconf ip
> 
> edit after more testing i got
> 
> ```
> sh-4.4# /usr/sbin/wpa_supplicant -i ovs_eth70 -c /usr/syno/etc/wpa_supplicant.conf.ovs_eth70 -qq -B -P /var/run/wpa_supplicant.pid.ovs_eth70
> nl80211: Could not configure driver mode
> ovs_eth70: Failed to initialize driver interface
> 
> sh-4.4#/usr/syno/sbin/synonet --dhcp eth70
> synonet.c:214 Error: Invalid network interface [eth70]
> ```
> 
> so looks like wpa_supplicant dont like ovs_eth70 and synonet dont like eth70
> 
> to get it working i'm loading the modules with ifname=ovs_eth70 and ovs_eth70 as name in wpa_supplicant script

plz used eth8x， eth7x I left it to rndis

wjz304, 2025-02-06:
> 我在 25.1.4 更新后遇到问题
> 
> 更新后，如果我手动运行 bash -x /usr/bin/wireless_supplicant.sh eth70 ssid pass synonet 返回错误，我会得到 2 台设备 eth70 和 ovs_eth70
> 
> ```
> + /usr/syno/sbin/synonet --dhcp eth70
> synonet.c:214 Error: Invalid network interface [eth70]. 
> ```
> 
> 我尝试使用 ovs_eth70 并且情况略有好转，但我没有得到 IP，它卡在 zeroconf IP 上
> 
> 在我得到的更多测试后编辑
> 
> ```
> sh-4.4# /usr/sbin/wpa_supplicant -i ovs_eth70 -c /usr/syno/etc/wpa_supplicant.conf.ovs_eth70 -qq -B -P /var/run/wpa_supplicant.pid.ovs_eth70
> nl80211: Could not configure driver mode
> ovs_eth70: Failed to initialize driver interface
> 
> sh-4.4#/usr/syno/sbin/synonet --dhcp eth70
> synonet.c:214 Error: Invalid network interface [eth70]
> ```
> 
> 所以看起来 wpa_supplicant 不喜欢 ovs_eth70 和 Synonet 不喜欢 eth70
> 
> 为了让它工作，我在脚本中使用 ifname=ovs_eth70 加载模块ovs_eth70 wpa_supplicant

OK, maybe there is a logical deviation, I will check it

wjz304, 2025-02-06:
`/usr/syno/sbin/synonet --dhcp ovs_ethX` Any questions?

Kali-, 2025-02-15:
i think your latest changes is going in the right way, but i dont understand where wifi is loaded at boot as i get eth80 and the related ovs_eth80

i'm try to fix, but i cant understand where the modules is loaded as my load modules script on boot is 
```
insmod /lib/modules/cfg80211.ko
insmod /lib/modules/mac80211.ko 
insmod /lib/modules/8188eu.ko ifname=ovs_eth80

```
but i get eth80 and related ovs_eth80

anyway, i think if ovs is enabled, the wifi modules must be loaded only with ovs_ (ex: ovs_eth80) so the wifi extension is loaded on the right device and both wpa_supplicant and synonet will work

Kali-, 2025-02-15:
btw, i made PR for the latest changes

wjz304, 2025-02-17:
According to the test, it seems that the USB network card does not support open switch

I use pcie ax210 and can dhcp normally when OVS is turned on
I use USB aic880000 and cannot dhcp normally when OVS is turned on

wjz304, 2025-02-17:
```
cat /etc/sysconfig/network-scripts/ifcfg-eth80
cat /etc/sysconfig/network-scripts/ifcfg-ovs_eth80
```

wjz304, 2025-02-17:
https://github.com/RROrg/rr/releases/tag/25.2.1

Kali-, 2025-02-18:
Installed 25.2.1 and it works, i get both eth80 and ovs_eth80 

sh-4.4# cat /etc/sysconfig/network-scripts/ifcfg-eth80
DEVICE=eth80
ONBOOT=yes
BOOTPROTO=dhcp
IPV6INIT=dhcp
IPV6_ACCEPT_RA=1
BRIDGE=ovs_eth80
sh-4.4# cat /etc/sysconfig/network-scripts/ifcfg-ovs_eth80
DEVICE=ovs_eth80
ONBOOT=yes
BOOTPROTO=dhcp
IPV6INIT=dhcp
IPV6_ACCEPT_RA=1
PRIMARY=eth80
TYPE=OVS

wjz304, 2025-02-18:
yes, I fixed the problem that USB does not support ovs.
I am coding to support install DSM in wifi env

closed 2025-02-19 by wjz304