RROrg/rr #5457: BUG

nicolas-ncc, 2025-01-17:
请填写以下信息.  
Please fill in the following information.  

Install ENV: (You can find it in the boot interface.)  
 * DMI:  Windows 11 Chrome
 * CPU:  Intel i9 9900T 
 * NIC:  Virtio (PVE NIC)

RR version: (You can find it in the update menu.)  
 * RR:  latest
 * addons:  
 * modules:  
 * lkms:  

DSM: 
 * model:  all
 * version:  all

Issue:  Unable to connect to Synology website, please check the network and try again, or use 'Parse Pat'!
I can confirm network connection is fine as I config it on RR web interface.
Already tried upload pat file to /tmp/pats folder can start build loader and start install but will fail when 99% installation with network issue.

[image]

logs:  

(## 因为 log中存在 SN/MAC 等一些敏感信息, 当提供完整文件时请自行抹除他们, 当然你也可以发送到我的邮箱. ##)  
(## Because the log contains some sensitive information such as SN/MAC, please delete them when providing the complete file. Of course, you can also send it to my email. ##)    
...

(请先看一下#173、#175、#226 的内容)  
(Plz review the content of #173, #175, #226 first)  
...

(如果你只是说 XXX 不能用, 什么详细信息也不提供, 我也只能说感谢你的反馈.)  
(If you just say XXX doesn't work without providing any details, I can only say thank you for your feedback.)  
...

wjz304, 2025-01-17:
https://github.com/RROrg/rr/releases/tag/25.1.2

wjz304, 2025-01-17:
plz upload the pat in  install

closed 2025-01-17 by wjz304

nicolas-ncc, 2025-01-17:
Its can work by sulotion 3, thnaks for help!!!