RROrg/rr #5483: SSD缓存硬盘状态报错“已检测到”

Ryanznoco, 2025-01-18:
ssd硬盘红色状态为已检测到，存储池这里又显示缓存组缺失（没有移除缓存组的入口），也没有在线重组按钮，新的m.2硬盘也没有添加成为缓存组的选项，这种情况应该怎么办呢？

[image]

[image]

[image]

Ryanznoco, 2025-01-18:
Install ENV: (You can find it in the boot interface.)  
 * DMI:  
 * CPU:  E3-1204L-v5
 * NIC:  (pid & vid)

RR version: (You can find it in the update menu.)  
 * RR:  
 * addons:  
 * modules:  
 * lkms:  

DSM: 
 * model:  DVA1622
 * version:  DSM 7.2.2-72806 Update 2

Issue:  
ssd硬盘红色状态为已检测到，存储池这里又显示缓存组缺失（没有移除缓存组的入口），也没有在线重组按钮，新的m.2硬盘也没有添加成为缓存组的选项

logs:

hiby-arch, 2025-03-26:
same issue。每次重启之后就会变成这样，无法处理。

wjz304, 2025-05-08:
尝试 添加cmdline 中的  pcie_port_pm=off  参数

closed 2025-09-07 by wjz304