RROrg/rr #5756: Would it be possible to activate ethernet Intel I217-LM drivers ?

spammoogle, 2025-01-28:
Install ENV: (You can find it in the boot interface.)

DMI: Dell Inc. Optiplex 9020/00V62H, BIOS A25 05/30/20219
CPU: Intel (R) Core (TM) i5-4590 CPU 3,30 GHz
NIC: (pid & vid)
vid: "0x058f"
pid: "0x6387"

[image]

RR version: (You can find it in the update menu.)

- RR: 23.11.1
- addons: attach_image_file
- modules: all
- lkms: 

DSM: 
- model: DS1621+
- version: DSM 7.2.2-72806 Update 2
 
Issue: 
Activate ethernet Intel I217-LM drivers

logs:
How can I get it?

PeterSuh-Q3, 2025-01-31:
The Linux driver for I217-LM is already built into rr's default module as an e1000e module.
It seems that e1000e could not have not worked.
However, for some reason, a kernel panic may occur and the nic may stop responding.
Fortunately, your workstation provides a COM1 serial port.

You can use a Serial to USB connector on this port to check the log to see what causes the kernel panic.
Monitoring via the serial port is always possible regardless of whether the nic is enabled or not.

https://www.aliexpress.com/item/1005006239265009.html

[image]

Which Synology models have you tried?

spammoogle, 2025-02-01:
I'll try a few more models before trying to detect the kernel panic via the COM port.

Thank you very much.

models tried:
DS1621+
DS620slim

wjz304, 2025-02-17:
Please provide the pid vid of the network card

spammoogle, 2025-02-19:
VID (Vendor ID): 8086 (Intel Corporation).

PID (Device ID): 15b7

wjz304, 2025-02-19:
```
root@ING-PC:/v1000-4.4.302# modinfo e1000e.ko | grep 15B7
alias:          pci:v00008086d000015B7sv*sd*bc*sc*i*
```
support, 
If you think it has no driver, please use other methods such as USB network card to confirm

closed 2025-02-20 by wjz304