RROrg/rr #6050: BUG:disk/disk_is_mv1475_driver.c:66 Can't get sata chip name from pattern

gowcar, 2025-02-04:
I met the same issue with [issue 1228](https://github.com/RROrg/rr/issues/1228)
May I know any updates on this issue?

I installed DSM on PVE 6.0
the SCSI controller is: Broadcom / LSI SAS2308 PCI-Express Fusion-MPT SAS-2
model: SA6400
DSM version: DSM 7.2.2-72806 Update 2
RR image version: 25.1.4

in the /var/log/messags, output this error message every 5 minutes:

`
2025-02-04T17:13:28+08:00 HomeNAS03 synostgd-disk[3408]: disk/disk_is_mv_soc_driver.c:71 Can't get sata chip name from pattern /sys/block/sata3/device/../../scsi_host/host*/proc_name
2025-02-04T17:13:28+08:00 HomeNAS03 synostgd-disk[3408]: disk/disk_is_mv1475_driver.c:66 Can't get sata chip name from pattern /sys/block/sata3/device/../../scsi_host/host*/proc_name
2025-02-04T17:13:28+08:00 HomeNAS03 synostgd-disk[3413]: disk/disk_is_mv_soc_driver.c:71 Can't get sata chip name from pattern /sys/block/sata4/device/../../scsi_host/host*/proc_name
2025-02-04T17:13:28+08:00 HomeNAS03 synostgd-disk[3413]: disk/disk_is_mv1475_driver.c:66 Can't get sata chip name from pattern /sys/block/sata4/device/../../scsi_host/host*/proc_name
2025-02-04T17:13:28+08:00 HomeNAS03 synostgd-disk[3415]: disk/disk_is_mv_soc_driver.c:71 Can't get sata chip name from pattern /sys/block/sata5/device/../../scsi_host/host*/proc_name
2025-02-04T17:13:28+08:00 HomeNAS03 synostgd-disk[3415]: disk/disk_is_mv1475_driver.c:66 Can't get sata chip name from pattern /sys/block/sata5/device/../../scsi_host/host*/proc_name
2025-02-04T17:13:28+08:00 HomeNAS03 synostgd-disk[3417]: disk/disk_is_mv_soc_driver.c:71 Can't get sata chip name from pattern /sys/block/sata6/device/../../scsi_host/host*/proc_name
2025-02-04T17:13:28+08:00 HomeNAS03 synostgd-disk[3417]: disk/disk_is_mv1475_driver.c:66 Can't get sata chip name from pattern /sys/block/sata6/device/../../scsi_host/host*/proc_name
2025-02-04T17:13:28+08:00 HomeNAS03 synostgd-disk[3419]: disk/disk_is_mv_soc_driver.c:71 Can't get sata chip name from pattern /sys/block/sata7/device/../../scsi_host/host*/proc_name
2025-02-04T17:13:28+08:00 HomeNAS03 synostgd-disk[3419]: disk/disk_is_mv1475_driver.c:66 Can't get sata chip name from pattern /sys/block/sata7/device/../../scsi_host/host*/proc_name
`

gowcar, 2025-02-05:
RR: 25.1.4
DSM: DSM 7.2.2-72806 Update 2
model: SA6400
RR image version: 25.1.4

Issue:
**Error messages keep occurs in /var/log/messages**

logs:
`
2025-02-05T15:02:05+08:00 HomeNAS03 synostgd-disk[16481]: disk/disk_is_mv_soc_driver.c:71 Can't get sata chip name from pattern /sys/block/sata3/device/../../scsi_host/host*/proc_name
2025-02-05T15:02:05+08:00 HomeNAS03 synostgd-disk[16481]: disk/disk_is_mv1475_driver.c:66 Can't get sata chip name from pattern /sys/block/sata3/device/../../scsi_host/host*/proc_name
2025-02-05T15:02:05+08:00 HomeNAS03 synostgd-disk[16485]: disk/disk_is_mv_soc_driver.c:71 Can't get sata chip name from pattern /sys/block/sata5/device/../../scsi_host/host*/proc_name
2025-02-05T15:02:05+08:00 HomeNAS03 synostgd-disk[16485]: disk/disk_is_mv1475_driver.c:66 Can't get sata chip name from pattern /sys/block/sata5/device/../../scsi_host/host*/proc_name
2025-02-05T15:02:05+08:00 HomeNAS03 synostgd-disk[16483]: disk/disk_is_mv_soc_driver.c:71 Can't get sata chip name from pattern /sys/block/sata4/device/../../scsi_host/host*/proc_name
2025-02-05T15:02:05+08:00 HomeNAS03 synostgd-disk[16483]: disk/disk_is_mv1475_driver.c:66 Can't get sata chip name from pattern /sys/block/sata4/device/../../scsi_host/host*/proc_name
2025-02-05T15:02:05+08:00 HomeNAS03 synostgd-disk[16487]: disk/disk_is_mv_soc_driver.c:71 Can't get sata chip name from pattern /sys/block/sata6/device/../../scsi_host/host*/proc_name
2025-02-05T15:02:05+08:00 HomeNAS03 synostgd-disk[16487]: disk/disk_is_mv1475_driver.c:66 Can't get sata chip name from pattern /sys/block/sata6/device/../../scsi_host/host*/proc_name
2025-02-05T15:02:05+08:00 HomeNAS03 synostgd-disk[16489]: disk/disk_is_mv_soc_driver.c:71 Can't get sata chip name from pattern /sys/block/sata7/device/../../scsi_host/host*/proc_name
2025-02-05T15:02:05+08:00 HomeNAS03 synostgd-disk[16489]: disk/disk_is_mv1475_driver.c:66 Can't get sata chip name from pattern /sys/block/sata7/device/../../scsi_host/host*/proc_name
`

wjz304, 2025-02-16:
https://github.com/RROrg/rr/releases/tag/25.2.0

closed 2025-02-16 by wjz304