RROrg/rr #6063: BUG: DX4600

vysecurity, 2025-02-05:
Please fill in the following information.  

Install ENV: (You can find it in the boot interface.)  
 * DMI:      UGREEN DX4600/To be filled by O.E.M, BIOS 5.19 12/21/2022
 * CPU:      Intel(R) Celeron(R) N5105 @ 2.00GHz
 * NIC:  (pid & vid)

RR version: (You can find it in the update menu.)  
 * RR:  RR v25.1.4 
 * addons:  all
 * modules:  all
 * lkms:  all

DSM: 
 * model:  7.2
 * version:  Latset

Issue:  

I'm trying to get it installed on a USB to boot into a DX4600. I had TCRP previously with DSM, so it was working fine, but I need to use RR to restore and install the bootloader for DSM so I can get back in. It detected fine (I can also access the web UI for RR), and I built the bootloader. Then, when accessing the bootloader, the DSM boots, but there is no IP address shown for DSM. It means there's no way to access the DSM instance.

[image]

[image]

[image]

[image]

[image]

The HDMI says: interface not operational. Wait a few minutes. Find DSM via HTTP://find.synology.com/ or Synology Assist.

I think there's no driver for network?

Thank you in advance for your help and comments.

wjz304, 2025-02-08:
N5105 建议安装 SA6400，其他型号尝试切换为直接驱动

vysecurity, 2025-02-09:
我也试了但是好像验证不到网卡

wjz304, 2025-02-09:
一般情况下， 这都不是网卡的问题。 内核崩溃在没有串口情况下很难调查。

翻译看看吧  https://rrorg.cn/archives/issues

vysecurity, 2025-02-09:
内核崩溃? 哪里看到的？

wjz304, 2025-02-20:
因为 RR 和 TCRP 的 base 新系统不一样，这在 kexec 时导致一些内存差别，
所以 我说 尝试切换为直接驱动，这将屏蔽 这些差别。

wjz304, 2025-02-20:
test https://github.com/RROrg/rr-build/actions/runs/13432156435

vysecurity, 2025-02-21:
This is 404. Not found. :( https://github.com/RROrg/rr-build/actions/runs/13432156435