RROrg/rr #6087: Required command /usr/bin/rr-loaderdisk.sh not found

liangjaden, 2025-02-08:
请填写以下信息.  
Please fill in the following information.  

Install ENV: (You can find it in the boot interface.)  
 * DMI:  
 * CPU:  
 * NIC:  (pid & vid)

RR version: (You can find it in the update menu.)  
 * RR:  25.1.3
 * addons:  
 * modules:  
 * lkms:  

DSM: 
 * model:  DVA1622
 * version:  7.2

Issue:  

安装rr-manager v2.0-79后插件菜单无法正常获取插件信息，接口返回Required command /usr/bin/rr-loaderdisk.sh not found

通过SSH登录查看/usr/bin目录文件，确实没有这个文件，在哪里可以下载恢复？

[image]

logs:  

(## 因为 log中存在 SN/MAC 等一些敏感信息, 当提供完整文件时请自行抹除他们, 当然你也可以发送到我的邮箱. ##)  
(## Because the log contains some sensitive information such as SN/MAC, please delete them when providing the complete file. Of course, you can also send it to my email. ##)    
...

(请先看一下#173、#175、#226 的内容)  
(Plz review the content of #173, #175, #226 first)  
...

(如果你只是说 XXX 不能用, 什么详细信息也不提供, 我也只能说感谢你的反馈.)  
(If you just say XXX doesn't work without providing any details, I can only say thank you for your feedback.)  
...

wjz304, 2025-02-08:
`ls /usr/rr/addons/`

liangjaden, 2025-02-10:
admin@DiskStation:~$ ls /usr/rr/addons/
notify.sh  nvmevolume.sh  rndis.sh  wol.sh

liangjaden, 2025-02-19:
@wjz304 可以再帮忙看下这个问题嘛？

wjz304, 2025-02-19:
`cat /usr/rr/backup/p1/user-config.yml | grep 'addons:' -A 20`

wjz304, 2025-02-19:
你自己没选则 mountloader addon

liangjaden, 2025-02-19:
@wjz304 抱歉没理解到，是需要重新编译引导吗？

wjz304, 2025-02-19:
nvmevolume 知道咋加吗？ 把 mountloader  加回来

liangjaden, 2025-02-20:
@wjz304 理解了，已解决，感谢~

closed 2025-02-20 by liangjaden