RROrg/rr #6429: nvme上有设备时，日志里会有类似nvme字眼的报错，会隔几秒写硬盘导致无法休眠

abcd1067350897, 2025-02-16:
请填写以下信息.  
Please fill in the following information.  

Install ENV: (You can find it in the boot interface.)  
 * DMI:  云星B250，OEM ITX-SC2/ITX-SC2, BIOS 5.12 08/12/2024
 * CPU:  Pentium(R) CPU G4560
 * NIC:  pid=0x1666 vid=0x0951

RR version: (You can find it in the update menu.)  
 * RR:  25.1.4
 * addons:  nvmesystem
 * modules:  默认
 * lkms:  默认

DSM: 
 * model:  DS918+ 
 * version:  7.2 69057

Issue:  
编译后nvme上有设备时，日志里会有类似nvme字眼的报错，会隔几秒写日志 硬盘导致无法休眠

我多次测试 使用U盘引导，不安装任何nvme硬盘仅安装sata硬盘时，正常
U盘引导 把系统安装进nvme硬盘内，有报错
nvme硬盘引导，安装进sata硬盘，有报错

麻烦您看一下，谢谢您！
logs:  
这是其中一部分scemd.log
2025-02-16T23:47:41+08:00 POI scemd[14331]: nvme_dev_port_check.c:29 Failed to get slot informtion of nvme0n1
2025-02-16T23:47:41+08:00 POI scemd[7299]: SYSTEM:  Last message 'nvme_dev_port_check.' repeated 4 times, suppressed by syslog-ng on POI
2025-02-16T23:47:41+08:00 POI scemd[14331]: nvme_dev_is_u2_slot.c:37 Failed to get slot informtion of nvme0n1
2025-02-16T23:47:41+08:00 POI scemd[14331]: disk/shared_disk_list_ops.c:24 Wrong parameter
2025-02-16T23:47:41+08:00 POI scemd[14331]: disk_temperature_update.c:53 Can't find the Disk Node whose path is /dev/nvme0n1
2025-02-16T23:47:43+08:00 POI scemd[14331]: nvme_dev_port_check.c:29 Failed to get slot informtion of nvme0n1
2025-02-16T23:47:43+08:00 POI scemd[14331]: fan/fan_table_type_disk_temperature_ops.c:130 no Disk List
2025-02-16T23:47:43+08:00 POI scemd[14331]: fan/fan_table_type_disk_temperature_ops.c:247 Invalid Disks temperature -1
2025-02-16T23:47:43+08:00 POI scemd[14331]: fan/fan_config_table_lookup.c:221  <disk_temperature> Temperature Get fail
2025-02-16T23:47:43+08:00 POI scemd[14331]: nvme_dev_port_check.c:29 Failed to get slot informtion of nvme0n1
2025-02-16T23:47:43+08:00 POI scemd[14331]: nvme_dev_port_check.c:29 Failed to get slot informtion of nvme0n1
2025-02-16T23:47:45+08:00 POI scemd[14331]: nvme_dev_port_check.c:29 Failed to get slot informtion of nvme0n1
2025-02-16T23:47:45+08:00 POI scemd[7299]: SYSTEM:  Last message 'nvme_dev_port_check.' repeated 3 times, suppressed by syslog-ng on POI
2025-02-16T23:47:45+08:00 POI scemd[14331]: fan/fan_table_type_disk_temperature_ops.c:130 no Disk List
2025-02-16T23:47:45+08:00 POI scemd[14331]: fan/fan_table_type_disk_temperature_ops.c:247 Invalid Disks temperature -1
2025-02-16T23:47:45+08:00 POI scemd[14331]: fan/fan_config_table_lookup.c:221  <disk_temperature> Temperature Get fail
2025-02-16T23:47:45+08:00 POI scemd[14331]: nvme_dev_port_check.c:29 Failed to get slot informtion of nvme0n1
2025-02-16T23:47:45+08:00 POI scemd[14331]: nvme_dev_is_u2_slot.c:37 Failed to get slot informtion of nvme0n1
2025-02-16T23:47:45+08:00 POI scemd[14331]: nvme_dev_port_check.c:29 Failed to get slot informtion of nvme0n1
2025-02-16T23:47:45+08:00 POI scemd[14331]: nvme_dev_port_check.c:29 Failed to get slot informtion of nvme0n1
2025-02-16T23:47:47+08:00 POI scemd[14331]: nvme_dev_port_check.c:29 Failed to get slot informtion of nvme0n1
2025-02-16T23:47:47+08:00 POI scemd[14331]: fan/fan_table_type_disk_temperature_ops.c:130 no Disk List
2025-02-16T23:47:47+08:00 POI scemd[14331]: fan/fan_table_type_disk_temperature_ops.c:247 Invalid Disks temperature -1
2025-02-16T23:47:47+08:00 POI scemd[14331]: fan/fan_config_table_lookup.c:221  <disk_temperature> Temperature Get fail
2025-02-16T23:47:47+08:00 POI scemd[14331]: nvme_dev_port_check.c:29 Failed to get slot informtion of nvme0n1
2025-02-16T23:47:47+08:00 POI scemd[14331]: nvme_dev_port_check.c:29 Failed to get slot informtion of nvme0n1
2025-02-16T23:47:47+08:00 POI scemd[14331]: nvme_dev_port_check.c:29 Failed to get slot informtion of nvme0n1
2025-02-16T23:47:47+08:00 POI scemd[7299]: SYSTEM:  Last message 'nvme_dev_port_check.' repeated 3 times, suppressed by syslog-ng on POI
2025-02-16T23:47:47+08:00 POI scemd[14331]: nvme_dev_is_u2_slot.c:37 Failed to get slot informtion of nvme0n1
2025-02-16T23:47:47+08:00 POI scemd[14331]: disk/shared_disk_list_ops.c:24 Wrong parameter
2025-02-16T23:47:47+08:00 POI scemd[14331]: disk_temperature_update.c:53 Can't find the Disk Node whose path is /dev/nvme0n1
2025-02-16T23:47:49+08:00 POI scemd[14331]: nvme_dev_port_check.c:29 Failed to get slot informtion of nvme0n1
2025-02-16T23:47:49+08:00 POI scemd[14331]: nvme_dev_port_check.c:29 Failed to get slot informtion of nvme0n1
2025-02-16T23:47:49+08:00 POI scemd[14331]: fan/fan_table_type_disk_temperature_ops.c:130 no Disk List
2025-02-16T23:47:49+08:00 POI scemd[14331]: fan/fan_table_type_disk_temperature_ops.c:247 Invalid Disks temperature -1
2025-02-16T23:47:49+08:00 POI scemd[14331]: fan/fan_config_table_lookup.c:221  <disk_temperature> Temperature Get fail
2025-02-16T23:47:49+08:00 POI scemd[14331]: nvme_dev_port_check.c:29 Failed to get slot informtion of nvme0n1
2025-02-16T23:47:49+08:00 POI scemd[14331]: nvme_dev_port_check.c:29 Failed to get slot informtion of nvme0n1
2025-02-16T23:47:51+08:00 POI scemd[14331]: nvme_dev_port_check.c:29 Failed to get slot informtion of nvme0n1
2025-02-16T23:47:51+08:00 POI scemd[7299]: SYSTEM:  Last message 'nvme_dev_port_check.' repeated 3 times, suppressed by syslog-ng on POI
2025-02-16T23:47:51+08:00 POI scemd[14331]: fan/fan_table_type_disk_temperature_ops.c:130 no Disk List
2025-02-16T23:47:51+08:00 POI scemd[14331]: fan/fan_table_type_disk_temperature_ops.c:247 Invalid Disks temperature -1
2025-02-16T23:47:51+08:00 POI scemd[14331]: fan/fan_config_table_lookup.c:221  <disk_temperature> Temperature Get fail
2025-02-16T23:47:51+08:00 POI scemd[14331]: nvme_dev_port_check.c:29 Failed to get slot informtion of nvme0n1
2025-02-16T23:47:53+08:00 POI scemd[14331]: nvme_dev_port_check.c:29 Failed to get slot informtion of nvme0n1
2025-02-16T23:47:53+08:00 POI scemd[14331]: fan/fan_table_type_disk_temperature_ops.c:130 no Disk List
2025-02-16T23:47:53+08:00 POI scemd[14331]: fan/fan_table_type_disk_temperature_ops.c:247 Invalid Disks temperature -1
2025-02-16T23:47:53+08:00 POI scemd[14331]: fan/fan_config_table_lookup.c:221  <disk_temperature> Temperature Get fail
2025-02-16T23:47:53+08:00 POI scemd[14331]: nvme_dev_port_check.c:29 Failed to get slot informtion of nvme0n1
2025-02-16T23:47:53+08:00 POI scemd[14331]: nvme_dev_port_check.c:29 Failed to get slot informtion of nvme0n1
2025-02-16T23:47:53+08:00 POI scemd[14331]: nvme_dev_port_check.c:29 Failed to get slot informtion of nvme0n1
2025-02-16T23:47:53+08:00 POI scemd[7299]: SYSTEM:  Last message 'nvme_dev_port_check.' repeated 4 times, suppressed by syslog-ng on POI
2025-02-16T23:47:53+08:00 POI scemd[14331]: nvme_dev_is_u2_slot.c:37 Failed to get slot informtion of nvme0n1
2025-02-16T23:47:53+08:00 POI scemd[14331]: disk/shared_disk_list_ops.c:24 Wrong parameter
2025-02-16T23:47:53+08:00 POI scemd[14331]: disk_temperature_update.c:53 Can't find the Disk Node whose path is /dev/nvme0n1
(## 因为 log中存在 SN/MAC 等一些敏感信息, 当提供完整文件时请自行抹除他们, 当然你也可以发送到我的邮箱. ##)  
(## Because the log contains some sensitive information such as SN/MAC, please delete them when providing the complete file. Of course, you can also send it to my email. ##)    
...

(请先看一下#173、#175、#226 的内容)  
(Plz review the content of #173, #175, #226 first)  
...

(如果你只是说 XXX 不能用, 什么详细信息也不提供, 我也只能说感谢你的反馈.)  
(If you just say XXX doesn't work without providing any details, I can only say thank you for your feedback.)  
...

abcd1067350897, 2025-02-16:
这个机器以前使用的之前Redpill引导（大概是这个名字？），当时是在网页编译，使用的是DS918+ 7.2.0-64561，当时应该没有遇到疯狂写硬盘的问题，现在整机原样升级25.1.4就遇到了这个问题
当时使用的引导
{"model":"DS918+","version":"7.2.0-64561","jun":"1","lkm":"dev","maxdisks":"8","maxlanport":"7","internalportcfg":"0xffff","netif_num":"2","vid":"0x0951","pid":"0x1666","diskidxmap":"06","sataportmap":"00","modules":"r8125"}

abcd1067350897, 2025-02-17:
感谢！

closed 2025-02-17 by abcd1067350897

wjz304, 2025-02-17:
并未屏蔽上述所有日志，也无法全部屏蔽

abcd1067350897, 2025-02-17:
目前换成了SA6400，好像没有这个问题了，nvme硬盘正常使用