RROrg/rr #6452: pve下安装rr引导后访问5000端口一直加载

JOYACEpoor, 2025-02-19:
请填写以下信息.  
Please fill in the following information.  

Install ENV: (You can find it in the boot interface.)  
 * DMI:  QEMU Standard PC (Q35 + ICH9, 2009), BIOS 3.20230228-4 06/06/2023
 * CPU:  qtj2 10750h(ES)
 * NIC:  (pid & vid)

[image]

RR version: (You can find it in the update menu.)  
 * RR:  25.2.1
 * addons:  
[image]

[image]

 * modules:  默认
 * lkms:  prod

DSM: 
 * model:  918+
 * version:  7.2

[image]

Issue:  pve下安装rr引导后访问5000端口一直加载，无法进行下一步,want help

[image]

logs:  

[logs.txt](https://github.com/user-attachments/files/18860675/logs.txt)

wjz304, 2025-02-19:
[image]

JOYACEpoor, 2025-02-19:
[image]

JOYACEpoor, 2025-02-19:
可以搜到群辉，但是访问ip:5000还是一直转圈，试过sa6400也是一样，any advice?
[image]

JOYACEpoor, 2025-02-19:
发现了，需要关闭代理，虽然代理用clash分流了，但是仍然有影响

closed 2025-02-19 by JOYACEpoor