RROrg/rr #6457: BUG 天钡 R1，RR25.1.1 开始无法进入引导

lrw514, 2025-02-19:
请填写以下信息.  
Please fill in the following information.  

Install ENV: (You can find it in the boot interface.)  
 * DMI:  
 * CPU:   N100
 * NIC:  (pid & vid)

RR version: (You can find it in the update menu.)  
 * RR:   25.1.0 good, 25.1.1 not working.
 * addons:  
 * modules:  
 * lkms:  

DSM: 
 * model:  SA6400
 * version:  7.2.1

Issue:  RR 25.1.0 版本可以进入 RR，也可以正常引导 DSM。 从 25.1.1 版本开始，无法进入 RR，开机闪现 RR logo 后即进入 BIOS 界面。此时回退 25.1.0 依然正常。
比对日志，发现有 NVME和 UEFI 两处关键改动，在R1， NVME 检测依然为 0， 未变化
暂时无法理解 UEFI 改动为什么会对 grub 引导产生影响。

logs:  

...

wjz304, 2025-02-20:
我的n100 正常，啥信息都没有，没法调查，方便就 qq我 远程吧，不方便就随缘吧

wjz304, 2025-02-20:
test https://github.com/RROrg/rr-build/actions/runs/13432156435

lrw514, 2025-02-20:
> test https://github.com/RROrg/rr-build/actions/runs/13432156435

rr-manager 提示 没有 PR_VERSION
rr 更新提示 checksum 不匹配： 
324fb044e7ca35a16c003fefebaa4efd  ./updateall.zip

wjz304, 2025-02-20:
updateall.zip 要解压一层的

lrw514, 2025-02-20:
> updateall.zip 要解压一层的

重启，显示 AME bios 的红图标，然后 右下角 A1 A2 -> A9 快速跳变，直接进入 bios 设置界面

wjz304, 2025-02-20:
grub 进不去？

lrw514, 2025-02-20:
进不去，直接进到 bios 了，看起来像是 grub  load 失败。 没串口猜不到哪里错了

wjz304, 2025-02-20:
引导盘第一分区 boot/grub/grubenv 文件 看看

wjz304, 2025-02-20:
环境信息

lrw514, 2025-02-20:
linux_gfx_mode=keep
若干个#

wjz304, 2025-02-20:
keep 改成 text

lrw514, 2025-02-20:
依旧出错，右下角的数字变为 B4

wjz304, 2025-02-20:
boot/grub/grub.cfg

删除下面这部分
[image]

lrw514, 2025-02-20:
good 

> boot/grub/grub.cfg
> 
> 删除下面这部分 [image]

keep 改成 text

可以进入 RR，但不会自动进入 DSM

lrw514, 2025-02-20:
withefi earlyprintk syno_hw_version=SA6400 console=ttyS0,115200n8 consoleblank=600 split_lock_detect=off netif_num=2 pid=0x5583 earlycon=uart8250,io,0x3f8,115200n8 mac2=021132251BF6 mac1=021132251BF5 sn=VHOQY7ZFK55NN modprobe.blacklist=evbug,cdc_ether pcie_aspm=off rootwait vid=0x0781 loglevel=15 HddHotplug=1 skip_vender_mac_interfaces=0,1 panic=5 vender_format_version=2 log_buf_len=32M root=/dev/md0 syno_ttyS1=serial,0x2f8 syno_ttyS0=serial,0x3f8 mev=physical
grub-editenv: error: invalid environment block.

lrw514, 2025-02-20:
> > good
> > > boot/grub/grub.cfg
> > > 删除下面这部分 [image]
> > 
> > 
> > keep 改成 text
> > 可以进入 RR，但不会自动进入 DSM
> 
> 到底是 删了可以？ 还是 keep 改成 text 可以？

目前是两个都改了

wjz304, 2025-02-20:
把grubenv 哪个文件删了

lrw514, 2025-02-20:
删除 grubenv，且取掉 grub.cfg 那一段后，可以正常启动了

wjz304, 2025-02-20:
你的主板 fwsetup 不支持参数，导致直接进bios

lrw514, 2025-02-20:
更糟糕，fwsetup 都不支持
不加参数也是进入 bios

wjz304, 2025-02-20:
等下个版本

wjz304, 2025-02-21:
https://github.com/RROrg/rr/releases/tag/25.2.3