RROrg/rr #6729: BUG

DestoryG, 2025-02-26:
请填写以下信息.  
Please fill in the following information.  

Install ENV: (You can find it in the boot interface.)  
 * DMI:  Founder Computer iFound T229/Mainboard 2980UD3 VA, BIOS 5.011 07/30/2019
 * CPU:  Celeron 2980U
 * NIC:  rtl8168

RR version: (You can find it in the update menu.)  
 * RR:  25.2.3
 * addons: 25.2.3
 * modules:  25.2.3
 * lkms:  25.1.0

DSM: 
 * model:  DS918+
 * version:  7.2.2 72806

Issue:  安装PAT时，提示文件无法安装，已经毁损，尝试过其他型号以及版本，在线离线安装DSM都一样的提示

logs:  

(## 因为 log中存在 SN/MAC 等一些敏感信息, 当提供完整文件时请自行抹除他们, 当然你也可以发送到我的邮箱. ##)  
(## Because the log contains some sensitive information such as SN/MAC, please delete them when providing the complete file. Of course, you can also send it to my email. ##)    
...

[image]

(请先看一下#173、#175、#226 的内容)  
(Plz review the content of #173, #175, #226 first)  
...

(如果你只是说 XXX 不能用, 什么详细信息也不提供, 我也只能说感谢你的反馈.)  
(If you just say XXX doesn't work without providing any details, I can only say thank you for your feedback.)  
...

DestoryG, 2025-02-26:
勾选直接启动后成功安装启动，原因依然不明

wjz304, 2025-03-11:
现有信息不足以调查

closed 2025-03-11 by wjz304