RROrg/rr #6748: Question  - Just saw the release notes for v25.2.4.

eragon4k, 2025-02-27:
Does this mean we cannot use RRmanager to upgrade to v25.2.4?

Also, can you elaborate a little more on the update process?

Thank you.

/////////////////////////////
⚠️⚠️⚠️Warning:
Incompatible upgrade due to the newly added configuration items.

Upgrade countermeasure 1: Execute touch /mnt/p1/.upgraded in RR shell # Execute before upgrading or before compiling after upgrading.
Upgrade countermeasure 2: Upgrade twice this version in a row and then compile and boot.
Upgrade countermeasure 3: After upgrading, perform an "Try to recovery a installed DSM system" operation and then compile and boot.

wjz304, 2025-02-28:
1. 先更新 到 25.2.4 
[image]

2. 重启到 RR 后再次强制更新 25.2.4

[image]

3. 重启到RR 后在编译启动

eragon4k, 2025-02-28:
First, thank you for your fast response.

So, RR manager is not an option with this version?

wjz304, 2025-02-28:
Do not use RR manager for upgrades for this time. This version change is also to fix some upgrade defects of RR manager.

eragon4k, 2025-02-28:
Got it. Thank you.

closed 2025-02-28 by eragon4k

eragon4k, 2025-03-01:
> Do not use RR manager for upgrades for this time. This version change is also to fix some upgrade defects of RR manager.

After this, can we return to using the RR manager for updates in future releases?