RROrg/rr #6752: BUG update v25.2.4 failed

JaWaoo, 2025-02-28:
esxi 

update v25.2.3 to v25.2.4 failed

[image]

wjz304, 2025-02-28:
```
ls -al /mnt/p1

grep ^pat /mnt/p1/user-config.yml
grep ^model /mnt/p1/user-config.yml
```

JaWaoo, 2025-02-28:
> ```
> ls -al /mnt/p1
> 
> grep ^pat /mnt/p1/user-config.yml
> grep ^model /mnt/p1/user-config.yml
> ```

[image]

wjz304, 2025-02-28:
[image]

JaWaoo, 2025-02-28:
> [image]

 thank you

closed 2025-02-28 by JaWaoo