RROrg/rr #6833: BUG - run RR 25.2.2 fine, but i reinstall 25.2.3 or 25.2.4 but DSM dont load

vipkattkool, 2025-03-02:
请填写以下信息.  
Please fill in the following information.  

Install ENV: (You can find it in the boot interface.)  
 * DMI:  TUF B365M-PLUS GAMING, BIOS 2208 07/09/2021
 * CPU:  Intel(R) Core(TM) i3-9100 CPU
 * NIC: Intel I219-V: e1000e (8086:15b8), Intel I225-V: igc (8086:15f3)

RR version: (You can find it in the update menu.)  
 * RR:  25.2.4
 * Addons:  25.2.4
 * Modules:  25.2.3
 * LKMs:  25.1.0
 * CKs: 25.2.3

DSM: 7.2.2-72806-3
 * model:  SA6400
 * version:  7.2(72806)

Issue:
run RR 25.2.2 fine, but i reinstall 25.2.3 or 25.2.4 with same hardware, same config (MAC/SN, addon, module) it dont load into DSM

logs:

[image]

...

wjz304, 2025-03-05:
try "Direct boot: true"

vipkattkool, 2025-03-06:
ok, i will try in next weekend, i see you release new version too

wjz304, 2025-06-05:
If there is a problem with the latest version, please reopen it

closed 2025-06-05 by wjz304