RROrg/rr #7206: wireguard module

sergejostir, 2025-03-07:
Hello.

I have wireguard module enabled in the loader, but I don't understand what does it do. Should it make wg command available in CLI?

Thanks

wjz304, 2025-03-11:
[image]

closed 2025-03-11 by wjz304

sergejostir, 2025-03-11:
@wjz304 What do you mean? That's a package you install manually on any Synology. So what's the role of wireguard module in RR loader then?

wjz304, 2025-03-12:
> [@wjz304](https://github.com/wjz304) What do you mean? That's a package you install manually on any Synology. So what's the role of the wireguard module in RR loader then?

The application depends on the driver. Neither party can use it alone. In fact, the application integrates the driver.

Earlier, the author of the application said that it was impossible to compile the driver for the 5.0 kernel and hoped that I would directly integrate the driver into the system.