RROrg/rr #7333: 关于安装DS920+版本7.2的引导加载失败问题

WKYZY, 2025-03-12:
作者你好！

去年我已经成功安装DS920+版本7.2并正常使用了差不多1年，期间并没有更新过。我不是24小时开机，每天都会开关一次，直到有一天开机我发现它突然访问不了了，我就尝试着重启，重启之后就可以访问了，但好景不长，两个周之后它就彻底挂了。我怀疑是引导出了问题，但无论我怎么重新安装引导，都访问不了。之后就一直搁置到现在。

最近我重新在pve上安装居然成功访问了，之前都是安装在物理机上，是同一台主机，点击恢复引导重新编译就正常访问了。但我手贱去点了更新，就又挂了。原来是7.2.2-72806 update 2，我看见网上说小版本可以直接更新，就想更新到7.2.2-72806 update 3，点了更新就彻底访问不了了，一直在自己重启。后来无论我怎么重新安装都卡在55%显示“无法安装 DSM：无法安装此文件，文件可能已经毁损。”

总是失败我索性就把硬盘都格式化了重新全新安装，但依旧失败卡55%。首先在编译完成启动后，进入群晖安装页面显示一排小字“Error: The bootloader disk is not successfully mounted, the installation will fail.”，继续点击安装就会在55%的时候弹窗“无法安装 DSM：无法安装此文件，文件可能已经毁损。”

后续我尝试安装DS920+版本7.1的也如此，最后我换了个SA6400版本7.2就正常安装上了，所以我怀疑对DS920+的引导编译存在bug。

[image]

[image]

[image]

[image]

wjz304, 2025-03-12:
https://rrorg.cn/archives/vmconfig

wjz304, 2025-03-12:
引导盘请使用sata

WKYZY, 2025-03-12:
啊，我应该早点来问的，刚刚我试了，没有再显示“Error: The bootloader disk is not successfully mounted, the installation will fail.”了，这肯定能成功安装了。之前我是跟着b站教程一步一步走的，其中没有提到引导盘要使用sata，现在自己弄了半天把硬盘都给格式化了得重新花时间拷贝数据了。

我可以问问那为啥SA6400不改sata也能正常安装使用呢？
然后就是之前一直正常使用的物理机DS920+没有更新过突然访问不了也是因为这个没有设置引导盘sata的问题吗？
最后就是推荐安装那个型号的群晖更稳定一些呢？

谢谢。