RROrg/rr #7336: help

DOK-26-RUS, 2025-03-12:
Is it possible to request, make support or add a driver for "SCU SATA" on the x9drw-if supermicro motherboard?

wjz304, 2025-03-17:
in RR shell
`lspci -nn`

DOK-26-RUS, 2025-03-17:
> в оболочке RR `lspci -nn`

[lspci -nn.txt](https://github.com/user-attachments/files/19290619/lspci.-nn.txt)

DOK-26-RUS, 2025-03-17:
> ls /sys/block/sd* -al

root@rr:/opt/rr# ls /sys/block/sd* -al
lrwxrwxrwx 1 root root 0 Mar 17 18:49 /sys/block/sda -> ../devices/pci0000:00/0000:00:1f.2/ata1/host0/target0:0:0/0:0:0:0/block/sda
lrwxrwxrwx 1 root root 0 Mar 17 18:49 /sys/block/sdb -> ../devices/pci0000:00/0000:00:1f.2/ata3/host2/target2:0:0/2:0:0:0/block/sdb
lrwxrwxrwx 1 root root 0 Mar 17 18:49 /sys/block/sdc -> ../devices/pci0000:00/0000:00:1f.2/ata4/host3/target3:0:0/3:0:0:0/block/sdc
lrwxrwxrwx 1 root root 0 Mar 17 18:49 /sys/block/sdd -> ../devices/pci0000:00/0000:00:1f.2/ata5/host4/target4:0:0/4:0:0:0/block/sdd
lrwxrwxrwx 1 root root 0 Mar 17 18:49 /sys/block/sde -> ../devices/pci0000:00/0000:00:1f.2/ata6/host5/target5:0:0/5:0:0:0/block/sde
lrwxrwxrwx 1 root root 0 Mar 17 18:49 /sys/block/sdf -> ../devices/pci0000:00/0000:00:11.0/0000:06:00.0/host6/port-6:0/end_device-6:0/target6:0:0/6:0:0:0/block/sdf
lrwxrwxrwx 1 root root 0 Mar 17 18:49 /sys/block/sdg -> ../devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1.1/2-1.1:1.0/host7/target7:0:0/7:0:0:0/block/sdg

wjz304, 2025-03-17:
> root@rr:/opt/rr# ls /sys/block/sd* -al lrwxrwxrwx 1 root root 0 Mar 17 18:49 /sys/block/sda -> ../devices/pci0000:00/0000:00:1f.2/ata1/host0/target0:0:0/0:0:0:0/block/sda lrwxrwxrwx 1 root root 0 [...]

sdb & sdc ?

wjz304, 2025-03-17:
> root@rr:/opt/rr# ls /sys/block/sd* -al lrwxrwxrwx 1 root root 0 Mar 17 18:49 /sys/block/sda -> ../devices/pci0000:00/0000:00:1f.2/ata1/host0/target0:0:0/0:0:0:0/block/sda lrwxrwxrwx 1 root root 0 [...]

Which drives are not shown in this result?

DOK-26-RUS, 2025-03-17:
[image]

[image]

DOK-26-RUS, 2025-03-17:
> > > ls /sys/block/sd* -al
> > 
> > 
> > root@rr:/opt/rr# ls /sys/block/sd* -al lrwxrwxrwx 1 root root 0 17 марта 18:49 /sys/block/sda -> ../devices/pci0000:00/0000:00:1f.2/ata1/host0/target0:0:0/0:0:0:0/block/sda lrwxrwxrwx 1 root root 0 17 марта 18:49 /sys/block/sdb -> ../devices/pci0000:00/0000:00:1f.2/ata3/host2/target2:0:0/2:0:0:0/block/sdb lrwxrwxrwx 1 root root 0 17 марта 18:49 /sys/block/sdc -> ../devices/pci0000:00/0000:00:1f.2/ata4/host3/target3:0:0/3:0:0:0/block/sdc lrwxrwxrwx 1 корень корень 0 17 марта 18:49 /sys/block/sdd -> ../devices/pci0000:00/0000:00:1f.2/ata5/host4/target4:0:0/4:0:0:0/block/sdd lrwxrwxrwx 1 корень корень 0 17 марта 18:49 /sys/block/sde -> ../devices/pci0000:00/0000:00:1f.2/ata6/host5/target5:0:0/5:0:0:0/block/sde lrwxrwxrwx 1 корень корень 0 17 марта 18:49 /sys/block/sdf -> ../devices/pci0000:00/0000:00:11.0/0000:06:00.0/host6/port-6:0/end_device-6:0/target6:0:0/6:0:0:0/block/sdf lrwxrwxrwx 1 корень корень 0 17 марта 18:49 /sys/block/sdg -> ../devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1.1/2-1.1:1.0/host7/target7:0:0/7:0:0:0/block/sdg
> 
> Какие диски не показаны в этом результате?

lrwxrwxrwx 1 root root 0 Mar 17 18:49 /sys/block/sdf -> ../devices/pci0000:00/0000:00:11.0/0000:06:00.0/host6/port-6:0/end_device-6:0/target6:0:0/6:0:0:0/block/sdf

wjz304, 2025-03-17:
> 06:00.0 Serial Attached SCSI controller [0107]: Intel Corporation C602 chipset 4-Port SATA Storage Control Unit [8086:1d6b] (rev 06)
> 
> yes, There is currently no driver for this device.

I will try to compile it, it may take some time, please pay attention to it in the next few days

DOK-26-RUS, 2025-03-17:
> 06:00.0 Контроллер последовательного интерфейса SCSI [0107]: набор микросхем Intel Corporation C602, 4-портовый блок управления накопителем SATA [8086:1d6b] (рев. 06)
> 
> да, в настоящее время для этого устройства нет драйвера.
ok, is it expected to be added? or can we not wait and leave it as it is?

wjz304, 2025-03-17:
> > 06:00.0 Контроллер последовательного интерфейса SCSI [0107]: набор микросхем Intel Corporation C602, 4-портовый блок управления накопителем SATA [8086:1d6b] (рев. 06)
> > да, в настоящее время для этого устройства нет драйвера.
> 
> ok, is it expected to be added? or can we not wait and leave it as it is?

Synology does not support this type of device, so the official does not provide the driver for this device.

wjz304, 2025-03-17:
[broadwellnk-isci-25.3.2.zip](https://github.com/user-attachments/files/19292840/broadwellnk-isci-25.3.2.zip)

/usr/lib/modules/isci.ko
/usr/lib/firmware/isci/isci_firmware.bin

DOK-26-RUS, 2025-03-17:
> [broadwellnk-isci-25.3.2.zip](https://github.com/user-attachments/files/19292840/broadwellnk-isci-25.3.2.zip)
> 
> upload to DSM /usr/lib/modules/isci.ko /usr/lib/firmware/isci/isci_firmware.bin
> 
> in DSM shell `insmod /usr/lib/modules/isci.ko`

insmod: ERROR: could not insert module /usr/lib/modules/isci.ko: Invalid module format

wjz304, 2025-03-17:
which DSM model?

DOK-26-RUS, 2025-03-17:
> which DSM model?

SA6400 DSM 7.2.2-72806

wjz304, 2025-03-17:
used epyc7002...

wjz304, 2025-03-18:
> > [epyc7002-isci-25.3.2.zip](https://github.com/user-attachments/files/19293283/epyc7002-isci-25.3.2.zip)
> > [broadwellnk-isci-25.3.2.zip](https://github.com/user-attachments/files/19292840/broadwellnk-isci-25.3.2.zip)
> > upload to DSM /usr/lib/modules/isci.ko /usr/lib/firmware/isci/isci_firmware.bin
> > in DSM shell `insmod /usr/lib/modules/isci.ko`
> 
> use epyc7002...

DOK-26-RUS, 2025-03-18:
> [epyc7002-isci-25.3.2.zip](https://github.com/user-attachments/files/19293283/epyc7002-isci-25.3.2.zip) [...]

insmod: ERROR: could not insert module /usr/lib/modules/isci.ko: Unknown symbol in module

wjz304, 2025-03-18:
[epyc7002-isci-25.3.3.zip](https://github.com/user-attachments/files/19319344/epyc7002-isci-25.3.3.zip)

upload to DSM
/usr/lib/modules/isci.ko
/usr/lib/modules/libsas.ko
/usr/lib/modules/scsi_transport_sas.ko
/usr/lib/firmware/isci/isci_firmware.bin

in DSM shell
```
insmod /usr/lib/modules/scsi_transport_sas.ko
insmod /usr/lib/modules/libsas.ko
insmod /usr/lib/modules/isci.ko
```

DOK-26-RUS, 2025-03-18:
[image]
everything applied without errors, but the 2TB hard drive did not appear.
The DSM was rebooting

wjz304, 2025-03-18:
> [image] everything applied without errors, but the 2TB hard drive did not appear. The DSM was rebooting

Do not reboot, because it is a test now, it is normal that the UI does not display.

After executing the above command, continue to execute `ls -al /sys/block/sata*`

DOK-26-RUS, 2025-03-18:
> After executing the above command, continue to execute ls -al /sys/block/sata*`

lrwxrwxrwx 1 root root 0 Mar 18 17:01 /sys/block/sata1 -> ../devices/pci0000:00/0000:00:1f.2/ata1/host0/target0:0:0/0:0:0:0/block/sata1
lrwxrwxrwx 1 root root 0 Mar 18 17:01 /sys/block/sata2 -> ../devices/pci0000:00/0000:00:1f.2/ata3/host2/target2:0:0/2:0:0:0/block/sata2
lrwxrwxrwx 1 root root 0 Mar 18 17:01 /sys/block/sata3 -> ../devices/pci0000:00/0000:00:1f.2/ata4/host3/target3:0:0/3:0:0:0/block/sata3
lrwxrwxrwx 1 root root 0 Mar 18 17:01 /sys/block/sata4 -> ../devices/pci0000:00/0000:00:1f.2/ata5/host4/target4:0:0/4:0:0:0/block/sata4
lrwxrwxrwx 1 root root 0 Mar 18 17:01 /sys/block/sata5 -> ../devices/pci0000:00/0000:00:1f.2/ata6/host5/target5:0:0/5:0:0:0/block/sata5

wjz304, 2025-03-18:
lsmod | grep isci

DOK-26-RUS, 2025-03-18:
> lsmod | grep isci

:~$ lsmod | grep isci
isci                  122880  0
libsas                 65536  1 isci
scsi_transport_sas     40960  2 isci,libsas

wjz304, 2025-03-18:
lspci -Qnnk

wjz304, 2025-03-18:
udevadm trigger --type=devices --action=add
  udevadm trigger --type=devices --action=change
  ls -al /sys/block/sata*

DOK-26-RUS, 2025-03-18:
> lspci -Qnnk

[lspci -Qnnk.txt](https://github.com/user-attachments/files/19321518/lspci.-Qnnk.txt)

wjz304, 2025-03-18:
> > lspci -Qnnk
> 
> [lspci -Qnnk.txt](https://github.com/user-attachments/files/19321518/lspci.-Qnnk.txt)

Unable to download

wjz304, 2025-03-18:
Let's wait for the next RR version to test.

DOK-26-RUS, 2025-03-18:
> which DSM model?

Yes, it's good

[lspci -Qnnk.txt](https://github.com/user-attachments/files/19321935/lspci.-Qnnk.txt)

wjz304, 2025-03-18:
https://github.com/RROrg/rr/releases/tag/25.3.3

wjz304, 2025-03-18:
upgrade this version, build and boot

wjz304, 2025-03-25:
> > upgrade this version, build and boot
> 
> Then, I'm ready.

Test results?

DOK-26-RUS, 2025-03-25:
> > > upgrade this version, build and boot
> > 
> > 
> > Then, I'm ready.
> 
> Test results?

Nothing has changed.

wjz304, 2025-03-25:
in dsm shell:

```
ls -al /usr/lib/modules/isci.ko
lsmod | grep isci
lspci -d ::107 -Qnnk
 ls -al /sys/block/sata*
```

DOK-26-RUS, 2025-03-25:
:~$ ls -al /usr/lib/modules/isci.ko
-rw-r--r-- 1 root root 245360 Mar 18 17:06 /usr/lib/modules/isci.ko

:~$ lsmod | grep isci
isci                  122880  0
libsas                 65536  1 isci
scsi_transport_sas     40960  2 isci,libsas

:~$ lspci -d ::107 -Qnnk
06:00.0 Serial Attached SCSI controller [0107]: Intel Corporation C602 chipset 4-Port SATA Storage Control Unit [8086:1d6b] (rev 06)
        Subsystem: Super Micro Computer Inc Device [15d9:0636]
        Kernel driver in use: isci

:~$ ls -al /sys/block/sata*
lrwxrwxrwx 1 root root 0 Mar 25 19:47 /sys/block/sata1 -> ../devices/pci0000:00/0000:00:1f.2/ata1/host0/target0:0:0/0:0:0:0/block/sata1
lrwxrwxrwx 1 root root 0 Mar 25 19:47 /sys/block/sata2 -> ../devices/pci0000:00/0000:00:1f.2/ata3/host2/target2:0:0/2:0:0:0/block/sata2
lrwxrwxrwx 1 root root 0 Mar 25 19:47 /sys/block/sata3 -> ../devices/pci0000:00/0000:00:1f.2/ata4/host3/target3:0:0/3:0:0:0/block/sata3
lrwxrwxrwx 1 root root 0 Mar 25 19:47 /sys/block/sata4 -> ../devices/pci0000:00/0000:00:1f.2/ata5/host4/target4:0:0/4:0:0:0/block/sata4
lrwxrwxrwx 1 root root 0 Mar 25 19:47 /sys/block/sata5 -> ../devices/pci0000:00/0000:00:1f.2/ata6/host5/target5:0:0/5:0:0:0/block/sata5

the connected hard drive is not displayed

wjz304, 2025-03-26:
> :~$ ls -al /usr/lib/modules/isci.ko -rw-r--r-- 1 root root 245360 Mar 18 17:06 /usr/lib/modules/isci.ko
>
> :~$ lsmod | grep isci isci 122880 0 libsas 65536 1 isci scsi_transport_sas 40960 2 [...]

`ls -al /dev/`

DOK-26-RUS, 2025-03-26:
> `ls -al /dev/`

[ls -al dev.txt](https://github.com/user-attachments/files/19462171/ls.-al.dev.txt)

wjz304, 2025-03-26:
ls -al /sys/block/iscsi1

DOK-26-RUS, 2025-03-26:
> ```
> ls -al /sys/block/iscsi1
> cat /sys/block/iscsi1/device/syno_block_info
> ```

SHD-home:~$ ls -al /sys/block/iscsi1
lrwxrwxrwx 1 root root 0 Mar 26 03:38 /sys/block/iscsi1 ->../devices/tcm_loop_0/tcm_loop_adapter_0/host8/target8:0:1/8:0:1:0/block/iscsi1
SHD-home:~$ cat /sys/block/iscsi1/device/syno_block_info
SHD-home:~$

wjz304, 2025-03-26:
```
dmesg | grep isci -A 20
```

DOK-26-RUS, 2025-03-26:
> ```
> dmesg | grep isci -A 20
> ```

[image]

DOK-26-RUS, 2025-03-29:
> It seems that the driver is not compatible. I don't have the device and cannot debug.

And if I can give you access, will it work?