RROrg/rr #7516: CPU and fan speed

Skyy404, 2025-03-15:
RR works great for me.  

What is really concern now - is there a way to control fan and cpu speed ? (i am on Terramaster 5095 Jasper Lake).

RR Manager somehow being installed but doesn't starts up.

Skyy404, 2025-03-15:
TerraMaster F2-423:
Processor Model Intel® Celeron® N5095 Jasper Lake
RR v25.3.2 DS920+(geminilake) 7.2(72806) 2150SBRZX24RD
no changes to RR

wjz304, 2025-03-17:
https://github.com/RROrg/rr/raw/main/docs/addons.xlsx

[image]

wjz304, 2025-03-17:
> RR Manager somehow being installed but doesn't starts up.

RRM is still under revision, please do not use it for the time being and wait for further notice

closed 2025-03-17 by wjz304