RROrg/rr #7518: Volume error

Skyy404, 2025-03-15:
How could i check the reasons for Volume becomes read-only.
Notice it for the second time. No errors. All works well after restart.
NVME in btfrs.

[image]

Skyy404, 2025-03-15:
any fast way to obtain this information?

so far i can add that possibly this is to misguided temp setup in RR.

log: 
disk_monitor.c:289 The temperature[38] of /dev/nvme0n1 >= T2. (T1: -273, T2: -273)
However, the thresholds are set to -273 (which is invalid and likely a configuration issue), and the actual temperature of the drive is 38°C.

i am using:
TerraMaster F2-423:
Processor Model Intel® Celeron® N5095 Jasper Lake
RR v25.3.2 DS920+(geminilake) 7.2(72806) 2150SBRZX24RD 

default loader settings,
only changed:

[image]

[image]

wjz304, 2025-03-18:
try add the hdddb addon.

wjz304, 2025-03-18:
support_fan=yes  目前不起作用

wjz304, 2025-03-18:
Are there any logs before and after `disk_monitor.c:289 The temperature[38] of /dev/nvme0n1 >= T2. (T1: -273, T2: -273)`?

I want to know which program outputs the logs

Skyy404, 2025-03-18:
> Are there any logs before and after `disk_monitor.c:289 The temperature[38] of /dev/nvme0n1 >= T2. (T1: -273, T2: -273)`?
> 
> I want to know which program outputs the logs

cat /var/log/messages | grep -i disk_monitor

[image]