RROrg/rr #7584: BUG在某些国产虚拟化平台中无法安装

etanman, 2025-03-18:
请填写以下信息.  
Please fill in the following information.  
因某些国产虚拟化平台无法选择硬盘类型，导致无法安装，能不能不支持所有磁盘类型？
Install ENV: (You can find it in the boot interface.)  
 * DMI:  
 * CPU:  
 * NIC:  (pid & vid)

RR version: (You can find it in the update menu.)  
 * RR:  
 * addons:  
 * modules:  
 * lkms:  

DSM: 
 * model:  
 * version:  

Issue:  

logs:  

(## 因为 log中存在 SN/MAC 等一些敏感信息, 当提供完整文件时请自行抹除他们, 当然你也可以发送到我的邮箱. ##)  
(## Because the log contains some sensitive information such as SN/MAC, please delete them when providing the complete file. Of course, you can also send it to my email. ##)    
...

(请先看一下#173、#175、#226 的内容)  
(Plz review the content of #173, #175, #226 first)  
...

(如果你只是说 XXX 不能用, 什么详细信息也不提供, 我也只能说感谢你的反馈.)  
(If you just say XXX doesn't work without providing any details, I can only say thank you for your feedback.)  
...

wjz304, 2025-03-18:
官方只支持 sata

etanman, 2025-03-19:
> 官方只支持 sata

那scsi接口为什么就可以安装呢？

wjz304, 2025-03-25:
能力有限，不能支持所有磁盘类型

closed 2025-05-01 by wjz304