RROrg/rr #7779: HyperV安装的群晖无法正常使用iSCSI服务

DYJ2217, 2025-03-25:
请填写以下信息.  
Please fill in the following information.  

Install ENV: (You can find it in the boot interface.)  
 * DMI:  
 * CPU:  
 * NIC:  (pid & vid)

RR version: (You can find it in the update menu.)  
 * RR:  
 * addons:  
 * modules: nvmesystem  
 * lkms:  

DSM: 
 * model:  SA6400
 * version:  7.2.2

Issue:  
客户端无法连接，群晖中显示正常
logs:  

(## 因为 log中存在 SN/MAC 等一些敏感信息, 当提供完整文件时请自行抹除他们, 当然你也可以发送到我的邮箱. ##)  
(## Because the log contains some sensitive information such as SN/MAC, please delete them when providing the complete file. Of course, you can also send it to my email. ##)    
...

(请先看一下#173、#175、#226 的内容)  
(Plz review the content of #173, #175, #226 first)  
...

(如果你只是说 XXX 不能用, 什么详细信息也不提供, 我也只能说感谢你的反馈.)  
(If you just say XXX doesn't work without providing any details, I can only say thank you for your feedback.)  
...

DYJ2217, 2025-03-25:
我在不同的主机进行测试都是同样的问题，期待解决！

wjz304, 2025-08-21:
https://github.com/RROrg/rr/issues/12945

Exia2929, 2025-12-30:
原來不是只有我掛不上嗎?不過NFS能掛PVE，如果能接受也是個選項。