RROrg/rr #7827: PVE VirtIO nic, DSM cannot open Jumbo frame(not able to set MTU > 1500)

soimy, 2025-04-02:
Install ENV: (You can find it in the boot interface.)  
 * DMI:  QEMU Standard PC Q35
 * CPU:  Intel Xeon E3-1265L v2 @ 2.50GHz
 * NIC:  (pid & vid) none (Used pve's VirtIO nic, confirmed module loaded

RR version: (You can find it in the update menu.)  
 * RR:  v25.3.2
 * addons:  
 * modules:  confirm virtio nic module loaded.
 * lkms:  

DSM: 
 * model:  3622xs+
 * version:  7.2(72806)

Issue:  
- PVE VirtIO nic, DSM cannot open Jumbo frame(not able to set MTU > 1500)
- can set MTU=9000 on vmxnet3/intel e1000e etc, but get only 1Gbps network, not 10Gbps.

wjz304, 2025-05-01:
The mtu setting of virtio_net is very complicated, you need to find PVE related information.
In addition, mnt will not change the rate,

[image]
[image]

wjz304, 2025-05-08:
无进一步反馈