RROrg/rr #7829: Hangs on "Booting..." after scheduling shutdown

AndreasFurster, 2025-04-02:
We're having an issue when booting after the system was shutdown by a schedule. 

It's stuck on "booting...". 

We can resolve this by removing some of the drives and try a reboot. Then the system starts normally. 

Then we manually shutdown the system using the user interface (not with a schedule). Then we can reconnect the removed drives and start without any problems. 

So the problem only occurs when using the schedule. We also tried running a custom task as root (`shutdown -P now`), but that causes the same issue.

AndreasFurster, 2025-04-04:
I think you misunderstood. We removed drives as in harddrives (hdd's). Not drivers.

wjz304, 2025-04-08:
OK, I will check it

wjz304, 2025-04-28:
你可以使用  `synowebapi --exec api=SYNO.Core.System method=shutdown version=2 force=false`  代替 `shutdown -P now`

wjz304, 2025-04-29:
If it doesn't work, please reopen

closed 2025-04-29 by wjz304

wjz304, 2025-04-29:
`/usr/syno/sbin/synopoweroff`