RROrg/rr #7968: BUG

tanlihng, 2025-04-12:
Install ENV: (You can find it in the boot interface.)  
 * DMI:  
 * CPU:  
 * NIC:  (pid & vid)

RR version: (You can find it in the update menu.)  
 * RR:  25.4.0
 * addons:  HDDdb、NvmeSystem、NvmeVolume
 * modules:  full
 * lkms:  

DSM: 
 * model:  SA6400
 * version:  7.2.1

Issue:  
DSM does not detect NVMe SSDs connected via ASM2812  (non-bifurcation M.2 adapter).
However, the drives are visible in the RR boot interface and via `ls /dev/nvme*` in SSH/Putty.
I suspect it may be caused by incorrect PCIe address mapping, resulting in DSM not recognizing the device correctly.

 Logs:
/dev/nvme0    /dev/nvme1    /dev/nvme1n1p1  /dev/nvme1n1p3  /dev/nvme-fabrics
/dev/nvme0n1  /dev/nvme1n1  /dev/nvme1n1p2  /dev/nvme1n1p5

root@SA6400:~# udevadm info /dev/nvme0n1
P: /devices/pci0000:00/0000:00:01.0/0000:01:00.0/0000:02:00.0/0000:03:00.0/nvme/nvme0/nvme0n1
N: nvme0n1
E: DEVNAME=/dev/nvme0n1
E: DEVPATH=/devices/pci0000:00/0000:00:01.0/0000:01:00.0/0000:02:00.0/0000:03:00.0/nvme/nvme0/nvme0n1
E: DEVTYPE=disk
E: MAJOR=259
E: MINOR=0
E: PHYSDEVBUS=pci
E: PHYSDEVDRIVER=nvme
E: PHYSDEVPATH=/devices/pci0000:00/0000:00:01.0/0000:01:00.0/0000:02:00.0/0000:03:00.0
E: SUBSYSTEM=block
E: SYNO_ATTR_SERIAL=0O1520000948
E: SYNO_DEV_DISKPORTTYPE=INVALID
E: SYNO_INFO_PLATFORM_NAME=epyc7002
E: SYNO_KERNEL_VERSION=5.10
E: SYNO_SUPPORT_USB_PRINTER=yes
E: SYNO_SUPPORT_XA=no
E: TAGS=:systemd:
E: USEC_INITIALIZED=903986

root@SA6400:~# udevadm info /dev/nvme1n1
P: /devices/pci0000:00/0000:00:01.0/0000:01:00.0/0000:02:0c.0/0000:06:00.0/nvme/nvme1/nvme1n1
N: nvme1n1
E: DEVNAME=/dev/nvme1n1
E: DEVPATH=/devices/pci0000:00/0000:00:01.0/0000:01:00.0/0000:02:0c.0/0000:06:00.0/nvme/nvme1/nvme1n1
E: DEVTYPE=disk
E: MAJOR=259
E: MINOR=1
E: PHYSDEVBUS=pci
E: PHYSDEVDRIVER=nvme
E: PHYSDEVPATH=/devices/pci0000:00/0000:00:01.0/0000:01:00.0/0000:02:0c.0/0000:06:00.0
E: SUBSYSTEM=block
E: SYNO_ATTR_SERIAL=AGIPEIWWK0203243
E: SYNO_DEV_DISKPORTTYPE=INVALID
E: SYNO_INFO_PLATFORM_NAME=epyc7002
E: SYNO_KERNEL_VERSION=5.10
E: SYNO_SUPPORT_USB_PRINTER=yes
E: SYNO_SUPPORT_XA=no
E: TAGS=:systemd:
E: USEC_INITIALIZED=906243
E: SYNO_KERNEL_VERSION=5.10
E: SYNO_SUPPORT_USB_PRINTER=yes
E: SYNO_SUPPORT_XA=no
E: TAGS=:systemd:
E: USEC_INITIALIZED=906243

wjz304, 2025-04-12:
```
cat /sys/block/nvme*/device/syno_block_info
synodisk --enum -t cache
```

tanlihng, 2025-04-13:
root@SA6400:~# cat /sys/block/nvme*/device/syno_block_info
pciepath=0000:00:01.0,00.0,00.0,00.0
pciepath=0000:00:01.0,00.0,0c.0,00.0

root@SA6400:~# synodisk --enum -t cache
root@SA6400:~# synodisk --enum -t cache
root@SA6400:~# synodisk --enum
Failed to enum disks
root@SA6400:~# synodisk --enum -t cache
root@SA6400:~# synodisk --enum cache
Failed to enum disks
root@SA6400:~# synodisk --enum -t cache
root@SA6400:~#

wjz304, 2025-04-13:
It seems that it is not compatible at the moment,
Need analysis

tanlihng, 2025-11-30:
Update: I successfully made DSM show the NVMe SSDs attached to the ASM2812 adapter using latest Synology_enable_M2_card v3.1.19 by 007revad.

Environment:
- RR 25.9.7
- DSM 7.2.2
- Model: SA6400
- ASM2812 non-bifurcation PCIe M.2 adapter
- Addons: HDDdb, NvmeSystem, NvmeVolume
- Modules: full

Fix / Workaround:
I used Dave Russell’s Synology_enable_M2_card script:
https://github.com/007revad/Synology_enable_M2_card

Steps:
1. Downloaded v3.1.19 "Source code (zip)".
2. Extracted to: /volume1/scripts/
3. Enabled SSH.
4. Ran the script and selected option **5) ALL**:

   sudo -s /volume1/scripts/syno_enable_m2_card.sh

After reboot:

✔ DSM now properly displays the NVMe SSDs in Storage Manager  
✔ synodisk works normally  
✔ NVMe volumes can be created without issue

wjz304, 2025-11-30:
cat /usr/syno/etc.defaults/adapter_cards.conf

tanlihng, 2025-11-30:
cat /usr/syno/etc.defaults/adapter_cards.conf
[FX2422N_sup_nic]
[FX2422N_sup_nvme]
FS6600N=yes
SA6400=yes
[FX2422N_sup_sata]
[E10M20-T1_sup_nic]
DS3617xsII=yes
RS3621rpxs=yes
DS2419+=yes
RS1219+=yes
RS18017xs+=yes
RS820+=yes
RS3618xs=yes
SA6400=yes
RS4025xs+=yes
RS4017xs+=yes
DS2419+II=yes
SA3410=yes
RS2818rp+=yes
SA3600=yes
RS3617rpxs=yes
DS3018xs=yes
RS818+=yes
DS1618+=yes
DS1817+=yes
RS1221rp+=yes
RS2423rp+=yes
DS2422+=yes
RS3617xs+=yes
RS2423+=yes
DS3617xs=yes
RS2421+=yes
DS1819+=yes
RS1224+=yes
DS3622xs+=yes
RS4021xs+=yes
RS822rp+=yes
RS818rp+=yes
RS1224rp+=yes
DS1517+=yes
RS1221+=yes
RS820rp+=yes
RS3621xs+=yes
SA3400=yes
SA6200=yes
RS2421rp+=yes
RS822+=yes
RS4023xs+=yes
RS2821rp+=yes
SA3610=yes
RS4024xs+=yes
[E10M20-T1_sup_nvme]
DS3617xsII=yes
RS3621rpxs=yes
DS2419+=yes
RS1219+=yes
RS18017xs+=yes
RS820+=yes
RS3618xs=yes
SA6400=yes
RS4025xs+=yes
RS4017xs+=yes
DS2419+II=yes
SA3410=yes
RS2818rp+=yes
SA3600=yes
RS3617rpxs=yes
DS3018xs=yes
RS818+=yes
DS1618+=yes
DS1817+=yes
RS1221rp+=yes
RS2423rp+=yes
DS2422+=yes
RS3617xs+=yes
RS2423+=yes
DS3617xs=yes
RS2421+=yes
DS1819+=yes
RS1224+=yes
DS3622xs+=yes
RS4021xs+=yes
RS822rp+=yes
RS818rp+=yes
RS1224rp+=yes
DS1517+=yes
RS1221+=yes
RS820rp+=yes
RS3621xs+=yes
SA3400=yes
SA6200=yes
RS2421rp+=yes
RS822+=yes
RS4023xs+=yes
RS2821rp+=yes
SA3610=yes
RS4024xs+=yes
[E10M20-T1_sup_sata]
DS3617xsII=yes
RS3621rpxs=yes
DS2419+=yes
RS1219+=yes
RS18017xs+=yes
RS820+=yes
RS3618xs=yes
SA6400=yes
RS4025xs+=yes
RS4017xs+=yes
DS2419+II=yes
SA3410=yes
RS2818rp+=yes
SA3600=yes
RS3617rpxs=yes
DS3018xs=yes
RS818+=yes
DS1618+=yes
DS1817+=yes
RS1221rp+=yes
RS2423rp+=yes
DS2422+=yes
RS3617xs+=yes
RS2423+=yes
DS3617xs=yes
RS2421+=yes
DS1819+=yes
RS1224+=yes
DS3622xs+=yes
RS4021xs+=yes
RS822rp+=yes
RS818rp+=yes
RS1224rp+=yes
DS1517+=yes
RS1221+=yes
RS820rp+=yes
RS3621xs+=yes
SA3400=yes
SA6200=yes
RS2421rp+=yes
RS822+=yes
RS4023xs+=yes
RS2821rp+=yes
SA3610=yes
RS4024xs+=yes
[M2D20_sup_nic]
[M2D20_sup_nvme]
DS3617xsII=yes
RS3621rpxs=yes
DS2419+=yes
RS18017xs+=yes
RS820+=yes
RS3618xs=yes
SA6400=yes
RS4025xs+=yes
RS4017xs+=yes
DS2419+II=yes
SA3410=yes
RS2818rp+=yes
SA3600=yes
RS3617rpxs=yes
DS3018xs=yes
DS1618+=yes
RS1221rp+=yes
RS2423rp+=yes
DS2422+=yes
RS3617xs+=yes
RS2423+=yes
DS3617xs=yes
RS2421+=yes
DS1819+=yes
RS1224+=yes
DS3622xs+=yes
RS4021xs+=yes
RS822rp+=yes
RS1224rp+=yes
RS2418rp+=yes
RS1221+=yes
RS820rp+=yes
RS3621xs+=yes
SA3400=yes
SA6200=yes
RS2421rp+=yes
RS822+=yes
RS4023xs+=yes
RS2821rp+=yes
SA3610=yes
RS2418+=yes
RS4024xs+=yes
[M2D20_sup_sata]
[M2D17_sup_nic]
[M2D17_sup_nvme]
[M2D17_sup_sata]
DS1817+=yes
DS1618+=yes
FS1018=yes
DS1517+=yes
RS2418rp+=yes
SA6400=yes
RS2818rp+=yes
DS3018xs=yes
RS2418+=yes
[M2D18_sup_nic]
[M2D18_sup_nvme]
RS4017xs+=yes
RS3617rpxs=yes
DS2419+II=yes
RS820+=yes
RS1224rp+=yes
RS822+=yes
RS820rp+=yes
RS822rp+=yes
RS818+=yes
DS1819+=yes
RS3618xs=yes
RS818rp+=yes
RS18017xs+=yes
RS1219+=yes
RS3617xs+=yes
RS2418rp+=yes
FS2017=yes
RS2818rp+=yes
RS1224+=yes
SA3400=yes
DS3617xsII=yes
RS1221rp+=yes
SA6400=yes
RS2418+=yes
DS1517+=yes
DS2419+=yes
DS1817+=yes
FS1018=yes
DS3617xs=yes
DS1618+=yes
DS3018xs=yes
RS1221+=yes
[M2D18_sup_sata]
RS4017xs+=yes
RS3617rpxs=yes
DS2419+II=yes
RS820+=yes
RS820rp+=yes
RS818+=yes
DS1819+=yes
RS3618xs=yes
RS818rp+=yes
RS18017xs+=yes
RS1219+=yes
RS3617xs+=yes
RS2418rp+=yes
FS2017=yes
RS2818rp+=yes
SA3400=yes
DS3617xsII=yes
SA6400=yes
RS2418+=yes
DS1517+=yes
DS2419+=yes
DS1817+=yes
FS1018=yes
DS3617xs=yes
DS1618+=yes
DS3018xs=yes

wjz304, 2025-11-30:
从结果看，似乎和 /usr/syno/etc.defaults/adapter_cards.conf 有关