RROrg/rr #7979: Driver Motorcomm YT6801 Gigabit Ethernet Adapter

Makc-Dem, 2025-04-13:
Please add Motorcomm YT6801 Gigabit Ethernet Adapter driver to the downloader.
device ID: 1F0A-6801 (rev 01)
System board intel AB2L, DualCore Intel Celeron N3350 (mini PC MLLSE M2 & MLLSE M2 AIR)

Makc-Dem, 2025-04-14:
Install ENV:
    DMI: Intel AB2L/ABL, BIOS 5.12 03/25/2024
    CPU: Intel(R) Celeron(R) CPU N3350 @ 1.10GHz
    NIC: (1f0a & 6801)

RR version:
    RR: 25.4.0
    addons: acpid,mountloader,powersched,reboottoloader,trivial,vmtools,cpuinfo,dbgutils,sensors,wireless
    modules: 8390,8139cp,8139too,9p,9pnet,9pnet_virtio,aacraid,acpi-cpufreq,acpi_call,adm1021,adm1031,adm9240,adt7470,adt7475,aic8800_fdrv,aic94xx,aic_load_fw,alx,apex,aqc111,asix,atl1c,atl1e,atlantic,auxiliary,ax88179_178a,b44,backlight,bcm-phy-lib,be2net,bitblit,bnx2,bnx2x,bnx2x_mod,bnxt_en,broadcom,btcoexist,button,cdc_ether,cdc_ncm,cdrom,cfbcopyarea,cfbfillrect,cfbimgblt,cfg80211,ch,ch341,cls_fw,cls_u32,cn,cp210x,cpufreq_conservative,cpufreq_governor,cpufreq_ondemand,cpufreq_performance,cpufreq_powersave,cpufreq_stats,cpufreq_userspace,crc-ccitt,crc-itu-t,cxgb,cxgb3,cxgb4,cxgb4vf,dm9601,dme1737,drm,drm_kms_helper,drm_panel_orientation_quirks,dvb-core,dvb-usb-dib0700,dvb-usb-dvbsky,dvb-usb,dvb_usb_v2,e1000,e1000e,eeprom_93cx6,efifb,efivarfs,ehci-hcd,ehci-pci,exfat,f71882fg,f75375s,fb,fb_sys_fops,fbcon,fbdev,font,ftdi_sio,gasket,gpio-beeper,gpio-f7188x,gpio-it87,gpio-sch,gpio-sch311x,gpio-vx855,hpilo,hpsa,hvc_console,hwmon-vid,i2c-algo-bit,i40e,i8042,i915,iavf,ib_addr,ib_cm,ib_core,ib_mad,ib_sa,ib_umad,ib_uverbs,ice,ide-core,ide-gd_mod,ide-pci-generic,igb,igbvf,igc,intel_auxiliary,intel_idle,iosf_mbi,ipv6,irqbypass,isci,iscsi_tcp,it87,iw_cm,iwldvm,iwlmvm,iwlwifi,ixgb,ixgbe,ixgbevf,jc42,jme,kvm-amd,kvm-intel,kvm,lib80211,libiscsi,libiscsi_tcp,libphy,libsas,llc,lm75,lm78,lm90,lm95245,m88ds3103,mac80211,mdio,megaraid,megaraid_mbox,megaraid_mm,megaraid_sas,mii,mlx4_core,mlx4_en,mlx4_ib,mlx5_core,mlx5_ib,mlxsw_core,mlxsw_pci,mmc_block,mmc_core,mpt3sas,mptbase,mptctl,mptsas,mptscsih,mptspi,mt7601u,mtk-sd,mvsas,mwifiex,mwifiex_sdio,mwifiex_usb,nct6683,nct6775,ne2k-pci,netxen_nic,ntfs,p8022,pcspeaker,pcspkr,phy-mt65xx-usb3,phy,piix,pktcdvd,pl2303,pmbus,pmbus_core,prism2_usb,processor,psnap,qcaspi,qed,qede,qla3xxx,qlcnic,qlge,r8125,r8125_txrx,r8126,r8126_txrx,r8152,r8168,r8168_tx,r8169,r8169_lk,raid_class,rc-core,realtek,regmap-i2c,rfkill-gpio,rfkill,rndis_host,rtc-cmos,rtl8187,rtl8188ee,rtl818x_pci,rtl8192c-common,rtl8192ce,rtl8192cu,rtl8192de,rtl8192ee,rtl8192se,rtl8723-common,rtl8723ae,rtl8723be,rtl8821ae,rtl8xxxu,rtl_pci,rtl_usb,rtlwifi,rtsx_pci,rtsx_pci_sdmmc,rtsx_usb,rtsx_usb_sdmmc,sch_htb,sch_netem,sch_sfq,scsi_transport_sas,scsi_transport_spi,sdhci-acpi,sdhci-pci-data,sdhci-pci,sdhci-pltfm,sdhci,sdio_uart,sfc,sg,si2157,si2168,skge,sky2,softcursor,sp2,sr_mod,ssb,staging,stp,synobios,syscopyarea,sysfillrect,sysimgblt,tehuti,tg3,thermal,tps65217,ts2020,tty0tty,uhci-hcd,usbnet,usbserial,ushc,vesafb,vfio,vga16fb,vgastate,via-rhine,via-sdmmc,via-velocity,video,virtio,virtio_blk,virtio_console,virtio_input,virtio_mmio,virtio_net,virtio_pci,virtio_ring,virtio_scsi,vmw_pvscsi,vmw_vmci,vmw_vsock_vmci_transport,vmxnet3,vsock,vub300,w83781d,w83793,wch,wch_pre,wireguard
    lkms: prod

DSM:
    model: DS920+
    version: 7.2.2-72806-0

Issue: Please add Motorcomm YT6801 Gigabit Ethernet Adapter driver to the downloader. (device ID: 1F0A-6801 (rev 01))

logs:

[rcExit.zip](https://github.com/user-attachments/files/19740512/rcExit.zip)

[jrExit.zip](https://github.com/user-attachments/files/19740515/jrExit.zip)

wjz304, 2025-04-17:
https://github.com/RROrg/rr-build/actions/runs/14492999488

test

Makc-Dem, 2025-04-22:
Works well

closed 2025-04-22 by Makc-Dem