RROrg/rr #8033: BUG

gasment, 2025-04-15:
请填写以下信息.  
Please fill in the following information.  

Install ENV: (You can find it in the boot interface.)  
 * DMI:  
 * CPU:  
 * NIC:  (pid & vid)

RR version: (You can find it in the update menu.)  
 * RR:  
 * addons:  
 * modules:  
 * lkms:  

DSM: 
 * model:  SA6400
 * version:  7.2.2

Issue:  
请求Coral Edge TPU PCIE版本的驱动支持。
https://github.com/[google/gasket-driver](https://github.com/google/gasket-driver)
官方文档https://coral.ai/docs/m2/get-started/
logs:  
03:00.0 System peripheral [0880]: Global Unichip Corp. Coral Edge TPU [1ac1:089a]
(## 因为 log中存在 SN/MAC 等一些敏感信息, 当提供完整文件时请自行抹除他们, 当然你也可以发送到我的邮箱. ##)  
(## Because the log contains some sensitive information such as SN/MAC, please delete them when providing the complete file. Of course, you can also send it to my email. ##)    
...

(请先看一下#173、#175、#226 的内容)  
(Plz review the content of #173, #175, #226 first)  
...

(如果你只是说 XXX 不能用, 什么详细信息也不提供, 我也只能说感谢你的反馈.)  
(If you just say XXX doesn't work without providing any details, I can only say thank you for your feedback.)  
...

wjz304, 2025-04-15:
gasket 不是加了么

gasment, 2025-04-15:
奥抱歉，忘记勾选模块了

gasment, 2025-04-16:
已勾选全部模块，重新编译引导后还是无驱动，手动insmod报错：
insmod: ERROR: could not insert module /usr/lib/modules/gasket.ko: Invalid parameter
insmod: ERROR: could not insert module /usr/lib/modules/apex.ko: Unknown symbol in module
lspci -v
06:00.0 Class 0880: Device 1ac1:089a (prog-if ff)
        Subsystem: Device 1ac1:089a
        Flags: fast devsel, IRQ 11
        Memory at 6000900000 (64-bit, prefetchable) [disabled] [size=16K]
        Memory at 6000800000 (64-bit, prefetchable) [disabled] [size=1M]
        Capabilities: [80] Express Endpoint, MSI 00
        Capabilities: [d0] MSI-X: Enable- Count=128 Masked-
        Capabilities: [e0] MSI: Enable- Count=1/32 Maskable- 64bit+
        Capabilities: [f8] Power Management version 3
        Capabilities: [100] Vendor Specific Information: ID=1556 Rev=1 Len=008 <?>
        Capabilities: [108] Latency Tolerance Reporting
        Capabilities: [110] L1 PM Substates
        Capabilities: [200] Advanced Error Reporting

gasment, 2025-04-16:
7410.070344] gasket: module uses symbol (dmabufbkpt_dma_buf_detach) from namespace DMA_BUF, but does not import it.
[ 7410.081609] gasket: Unknown symbol dmabufbkpt_dma_buf_detach (err -22)
[ 7410.088727] gasket: module uses symbol (dmabufbkpt_dma_buf_get) from namespace DMA_BUF, but does not import it.
[ 7410.099690] gasket: Unknown symbol dmabufbkpt_dma_buf_get (err -22)
[ 7410.106474] gasket: module uses symbol (dmabufbkpt_dma_buf_put) from namespace DMA_BUF, but does not import it.
[ 7410.117383] gasket: Unknown symbol dmabufbkpt_dma_buf_put (err -22)
[ 7410.124260] gasket: module uses symbol (dmabufbkpt_dma_buf_attach) from namespace DMA_BUF, but does not import it.
[ 7410.135559] gasket: Unknown symbol dmabufbkpt_dma_buf_attach (err -22)
[ 7410.142691] gasket: module uses symbol (dmabufbkpt_dma_buf_unmap_attachment) from namespace DMA_BUF, but does not import it.
[ 7410.154848] gasket: Unknown symbol dmabufbkpt_dma_buf_unmap_attachment (err -22)
[ 7410.162902] gasket: module uses symbol (dmabufbkpt_dma_buf_map_attachment) from namespace DMA_BUF, but does not import it.
[ 7410.174873] gasket: Unknown symbol dmabufbkpt_dma_buf_map_attachment (err -22)

gasment, 2025-04-16:
自己尝试编译了一下，成功加载了：
lsmod | grep gasket
gasket 86016 1 apex,[permanent], Live 0xffffffffa1350000 (OE)
dmabuf 61440 9 gasket,i915,drm_buddy,drm_display_helper,drm_kms_helper,ttm,drm,drm_mipi_dsi,i915_compat, Live 0xffffffffa011b000 (OE)
root@DSM-Server:/volume1/APP-Data/coral_driver/gasket-driver/src# lsmod | grep apex
apex 20480 0 - Live 0xffffffffa1366000 (OE)
gasket 86016 1 apex,[permanent], Live 0xffffffffa1350000 (OE)

大概是dma_buf模块的函数问题，dsm套了一个dmabufbkpt_*的封装，需要修改gasket的gasket_page_table.c来做兼容

wjz304, 2025-04-28:
已自行解决