RROrg/rr #8042: pve下SA6400存储管理器识别不到虚拟的scsi硬盘

rockft, 2025-04-17:
```
cat /sys/block/sata*/device/syno_block_info
pciepath=0000:00:1e.0,01.0,07.0
ata_port_no=0
driver=ahci
pciepath=0000:00:1e.0,01.0,07.0
ata_port_no=1
driver=ahci
pciepath=0000:00:1e.0,04.0,01.0
driver=virtio
```

syno_block_info中可以看到virtio 0000:00:1e.0,04.0,01.0 虚拟的scsi硬盘
但是dsm存储管理器中看不到
RR中模块已全部安装

手动编辑 dts 文件，导出的硬盘部分仅有 slot@1 的sata盘
```
{
	compatible = "Synology";
	model = "synology_epyc7002_sa6400";
	version = <0x01>;

	internal_slot@1 {
		protocol_type = "sata";

		ahci {
			pcie_root = "0000:00:1e.0,01.0,07.0";
			ata_port = <0x01>;
		};
	};
}
```

添加 slot@2 后，存储管理器中可以正常看到硬盘使用了
```
{
	compatible = "Synology";
	model = "synology_epyc7002_sa6400";
	version = <0x01>;

	internal_slot@1 {
		protocol_type = "sata";

		ahci {
			pcie_root = "0000:00:1e.0,01.0,07.0";
			ata_port = <0x01>;
		};
	};

	internal_slot@2 {
		protocol_type = "sata";

		virtio {
			pcie_root = "0000:00:1e.0,04.0,01.0";
		};
	};
}
```

原来在同样pve环境，老的引导(网上随便下的)和dsm918+上，添加虚拟scsi盘dsm可以自动识别使用
请问SA6400无法自动识别是软件限制还是BUG造成的？

### 附环境：

- **RR及DSM版本**
Welcome to RR v25.4.0   
 Boot Type: [UEFI] [SATA DoM]
型号:     SA6400(epyc7002)
版本:     7.2(72806)
内核:     official
LKM:      prod
MEV:      kvm
DMI:      QEMU Standard PC (Q35 + ICH9, 2009), BIOS 4.2025.02-3 04/03/2025
CPU:      12th Gen Intel(R) Core(TM) i3-12100
MEM:      7929 MB
Cmdline:
withefi earlyprintk syno_hw_version=SA6400 console=ttyS0,115200n8 consoleblank=600 split_lock_detect=off netif_num=2 pid=0x0001 earlycon=uart8250,io,0x3f8,115200n8 mac2=0011323884fe mac1=0011323884fd sn=2350W8RRWKR4W modprobe.blacklist=evbug,cdc_ether pcie_aspm=off rootwait nowatchdog vid=0x46f4 loglevel=15 HddHotplug=1 skip_vender_mac_interfaces=0,1 panic=5 vender_format_version=2 log_buf_len=32M root=/dev/md0 syno_ttyS1=serial,0x2f8 syno_ttyS0=serial,0x3f8 mev=kvm
插件: 未使用
模块: 全部安装

- **虚拟机配置**
PVE 8.4.1
内核 6.14.0-2-pve
[image]

wjz304, 2025-04-29:
不知道，没精力测试.

closed 2025-04-29 by wjz304

RhythmNz, 2025-11-25:
同样的问题，有人解决了吗